dotnet-bluetooth-le/dotnet-bluetooth-le #464: The Sample BLE.Client.Droid cannot Find Any Device

wagenheimer, 2020-10-16:
I'm executing the Android Demo from GitHub, but I could not find any Bluetooth devices. It seems to show me two errors:
Registration failed, unregister scannerId = 0
Scan failed, reason: app registration failed

## Steps to reproduce

1. Download the actual github version

2. Press Scan

## Expected behavior
Find near BluetoothLE devices.

## Actual behavior
"No devices are found", check the Output Window for Messages

### Crashlog

2020-10-15 22:28:34 [TRACE] (MvvmCross.Logging.MvxLog) Adapter >=21: Starting a scan for devices. ScanMode: LowLatency
10-15 22:28:34.004 I/mono-stdout(25114): 2020-10-15 22:28:34 [TRACE] (MvvmCross.Logging.MvxLog) Adapter >=21: Starting a scan for devices. ScanMode: LowLatency
10-15 22:28:34.006 D/BluetoothAdapter(25114): STATE_ON
10-15 22:28:34.009 I/chatty  (25114): uid=10423(com.ble.client) identical 1 line
10-15 22:28:34.010 D/BluetoothAdapter(25114): STATE_ON
10-15 22:28:34.011 D/BluetoothAdapter(25114): BLE support array set: 010011
10-15 22:28:34.011 D/BluetoothLeScanner(25114): Start Scan with callback
10-15 22:28:34.014 D/BluetoothLeScanner(25114): onScannerRegistered() - status=133 scannerId=0 mScannerId=0
10-15 22:28:34.015 D/BluetoothLeScanner(25114): Registration failed, unregister scannerId = 0
10-15 22:28:34.015 D/BluetoothLeScanner(25114): Scan failed, reason: app registration failed
2020-10-15 22:28:34 [TRACE] (MvvmCross.Logging.MvxLog) Adapter: Scan failed with code ApplicationRegistrationFailed
10-15 22:28:34.031 I/mono-stdout(25114): 2020-10-15 22:28:34 [TRACE] (MvvmCross.Logging.MvxLog) Adapter: Scan failed with code ApplicationRegistrationFailed
2020-10-15 22:28:44 [TRACE] (MvvmCross.Logging.MvxLog) Adapter: Scan timeout has elapsed.
2020-10-15 22:28:44 [TRACE] (MvvmCross.Logging.MvxLog) Adapter: Stopping the scan for devices.
2020-10-15 22:28:44 [TRACE] (MvvmCross.Logging.MvxLog) Adapter >= 21: Stopping the scan for devices.
10-15 22:28:44.021 I/mono-stdout(25114): 2020-10-15 22:28:44 [TRACE] (MvvmCross.Logging.MvxLog) Adapter: Scan timeout has elapsed.
10-15 22:28:44.023 I/mono-stdout(25114): 2020-10-15 22:28:44 [TRACE] (MvvmCross.Logging.MvxLog) Adapter: Stopping the scan for devices.
10-15 22:28:44.025 I/mono-stdout(25114): 2020-10-15 22:28:44 [TRACE] (MvvmCross.Logging.MvxLog) Adapter >= 21: Stopping the scan for devices.
10-15 22:28:44.027 D/BluetoothAdapter(25114): STATE_ON
10-15 22:28:44.030 D/BluetoothAdapter(25114): STATE_ON

## Configuration

**Version of the Plugin:** 
GitHub Most Recent Version

**Platform:** 
Android 10

**Device:**
Samsung Galaxy A71

wagenheimer, 2020-10-20:
For some unknown reason, it worked after I restarted the Phone.

closed 2020-10-20 by wagenheimer

CBRWLF98, 2020-11-17:
I had the same issue when working with bonding. As soon as I've bound the device, the scan stopped working like in this example. Setting the scan on higher timeout seemed to give it enough time to catch back on and scan some devices.

After restarting the smartphone everything worked like expected.

wagenheimer, 2021-01-27:
I still have this problem from time to time!
Restarting the Phone fix it 100%!

But it would be nice to have some way of detecting this error to show a dialog to the user!