dotnet-bluetooth-le/dotnet-bluetooth-le #468: How to get MAC address detected device

Denaindra, 2020-10-30:
## Steps to reproduce

1. Install Plugin 

2.Run the application

3.Detec devices

## Expected behavior
Get MAC to address from the device

## Actual behavior
Unable to get the mac address

### Crashlog
non

## Configuration

**Version of the Plugin:** 
 6.3.0.19

**Platform:** 
Android 10

**Device:** 
Xiomi 8 pro

xavi-dev, 2020-10-30:
Once you discover a device, you can get it from native device (only Android):

```c#
_bluetoothAdapter.DeviceDiscovered += async (adapter, args) =>
{
  IDevice device = args.Device;
  var macAddress = (string)device.NativeDevice.GetType().GetProperty("Address").GetValue(device.NativeDevice);
};
```

Denaindra, 2020-11-01:
Is there any way for iOS ?

xkampotx, 2020-11-03:
> Is there any way for iOS ?

iOS does not give you a possibility to get a mac address for a device.
https://stackoverflow.com/a/46236524/5927585

Auto72, 2020-11-13:
So, for iOS there is UUID available for each device, instead of the MAC-Address.
How can we read it?