dotnet-bluetooth-le/dotnet-bluetooth-le #470: Scanning with filter doesn't see any devices

Abesoddy, 2020-11-18:
I have an issue about the scan feature on Android 10 with a Samsung S20 Ultra. When I scanning devices with a service as filter, no device is discovered. However, the application has the Location permission and Bluetooth permission. I tested on Samsung galaxy Note 10 + and everything works... 

## Steps to reproduce

1. Call StartScanningForDevicesAsync() with a service in filter

```
var guid = Guid.Parse(Constants.dataTransfertServiceUUID);
await Adapter.StartScanningForDevicesAsync(new[] { guid }, _cancellationTokenSource.Token);
```

2. DeviceDiscovered is never called

## Expected behavior

Callback DeviceDiscovered is run whenever a device is found after calling StartScanningForDevicesAsync().

## Actual behavior

StartScanningForDevicesAsync() discovers no device.

### Crashlog

No exception is thrown.

## Configuration

**Version of the Plugin:** 2.1.1 and 2.2.0-pre2

**Platform:** Android 10

**Device:** Samsung galaxy S20 Ultra

buzzware, 2020-11-19:
See #454 and #466

Abesoddy, 2020-11-19:
Hi @buzzware 

Thanks for your reply, however the application has these permissions (AccessFineLocation and AccessCoarseLocation) (see the screenshots below).

[image]

[image]

Abesoddy, 2020-11-19:
And my function when I start the scan is this : 

       `/// <summary>
        /// Start scanning for BLE devices
        /// </summary>
        /// <param name="refresh"></param>
        public async void TryStartScanning(bool refresh = false)
        {
            if (Device.RuntimePlatform == Device.Android)
            {
                var status = await Permissions.CheckStatusAsync<Permissions.LocationAlways>();

                if (status != PermissionStatus.Granted)
                {
                    var permissionResult = await Permissions.RequestAsync<Permissions.LocationAlways>();

                    if (permissionResult != PermissionStatus.Granted)
                    {
                        string logMessage = "[BLEService, TryStartScanning] - Permission denied, not scanning";
                        System.Diagnostics.Debug.WriteLine(logMessage);
                        
                        AppInfo.ShowSettingsUI();
                        return;
                    }
                }
            }`

Abesoddy, 2020-11-27:
Anyone have this issue?

StephanRaab, 2021-01-25:
Just to confirm, the location permission is set to `always` in the app correct.
You have both Bluetooth permissions on in the manifest and when you scan without the filter you are able to find the device on the S20.

bebe84, 2024-10-01:
Hey, sorry to dig deep and re-discover this issue but it is happening for me also on Samung Galaxy A33 (Android 14). I have all permissions, location set to always and DiscoverDevices is never being called.

I can connect to the device directly with GUID but discovering it never works for me.

janusw, 2024-10-10:
> Hey, sorry to dig deep and re-discover this issue but it is happening for me also on Samung Galaxy A33 (Android 14). I have all permissions, location set to always and DiscoverDevices is never being called.

@bebe84 Do you use a scan filter? And if yes, how does it look like? I.e. how do you call `StartScanningForDevicesAsync`?

Are you using the latest version 3.1.0?

DeanZhuo, 2025-02-14:
im using android 14, version 3.1.0 with

`BleAdapter.ScanMode = ScanMode.Balanced;`
`BleAdapter.DeviceAdvertised += bleDeviceAdvertisedHandler;`
`BleAdapter.ScanTimeout = EffectBLE.BLE_SCAN_TIMEOUT;`
`ScanFilterOptions scanFilterOptions = new();`
`await BleAdapter.StartScanningForDevicesAsync(scanFilterOptions);`
and still no device