dotnet-bluetooth-le/dotnet-bluetooth-le #473: Are there plans to add support for canSendWriteWithoutResponse on iOS11+ ?

kwintAl, 2020-11-23:
On iOS devices, it's currently not possible to send a large amount of data to the peripheral device (without response).

On iOS11+ there is a property that indicates the current state: [canSendWriteWithoutResponse](https://developer.apple.com/documentation/corebluetooth/cbperipheral/2891512-cansendwritewithoutresponse?language=objc) and [peripheralIsReadyToSendWriteWithoutResponse](https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/2874034-peripheralisreadytosendwritewith?language=objc)

So in case canSendWriteWithoutResponse returns false, we should wait until the peripheralIsReadyToSendWriteWithoutResponse is called.

## Steps to reproduce

1. Connect to peripheral device

2. Send a large amount of data (without response)

## Expected behavior
no data loss

## Actual behavior
data loss

## Configuration

**Version of the Plugin:** e.g. 2.1.0

**Platform:**  iOS 14 / 13

**Device:** iPhone 11 / iPhone 6

geri-m, 2021-10-17:
The place where this check needs to be implemented is https://github.com/xabre/xamarin-bluetooth-le/blob/master/Source/Plugin.BLE.iOS/Characteristic.cs#L159. Currently the write would return immediately and the event from `peripheralIsReadyToSendWriteWithoutResponse` is not considered.

janusw, 2022-05-01:
Fixed with #554.

closed 2022-05-01 by janusw