dotnet-bluetooth-le/dotnet-bluetooth-le #475: IAdapter.StopScanningForDevicesAsync does not change IAdapter.IsScanning imediately

cdxybf, 2020-11-29:
## Steps to reproduce

1. Call await IAdapter.StartScanningForDevicesAsync(CancellationToken.None);

2. Call await IAdapter.StopScanningForDevicesAsync();

3. Refer IAdapter.IsScanning, it is still true.

## Expected behavior

 IAdapter.IsScanning should be false after await IAdapter.StopScanningForDevicesAsync() called.

## Actual behavior
It is still true until we use await Task.Delay(some milliseconds);

### Crashlog
N/A

## Configuration

**Version of the Plugin:** V2.1.1

**Platform:** Both Android and iOS

**Device:** Any

axa88, 2024-02-06:
v3.0.0/3.0.1 this issue is still valid

Checking the source, imho this never got fixed as the issue is a product of a bad design.
The scan understandably has to be stopped by a `CancelationToken`, but the way its implemented it isnt inducive to modification.

The API should have been designed with a `ScanStopped` event which would trigger whenever the scan has stopped regardless of the reason, and returned a custom `ScanStopEventArgs` indicating why the scan was stopped, which basically come down to timed out or manual.

Also `StopScanningForDevicesAsync` shouldn't have been made `async` at all as it runs sync anyway.  And the aforementioned even is triggerd to indicate it has actually been stopped.

axa88, 2024-02-11:
Instead of making a pull request to do more or less the same thing, this could fix the issue.

Create a function to start a scan, and use only it to start scanning..
All its doing here is setting a `Cancellation Token` to call the internal `Cancellation Token`, which in turn will stop the Scan when either stopped manually or timed out.  Do not call any `StartScanningForDevicesAsync` method directly.

```
	/// <summary>
	/// Not a task, no use to be
	/// </summary>
	/// <param name="timeout"> scan timeout.  With a default of 10s </param>
	private void ScanBegin(int timeout = 10000)
	{
		if (!CrossBluetoothLE.Current.Adapter.IsScanning)
		{
			CrossBluetoothLE.Current.Adapter.ScanTimeout = timeout + 1000; // set the internal timeout just a bit longer than desired timeout to prevent it from triggering internally
			_cancellationTokenSource = new((int)config.Timeout.TotalMilliseconds);
			_cancellationTokenSource.Token.Register(() => {	_cancellationTokenSource.Dispose(); });

			_adapter.StartScanningForDevicesAsync(cancellationToken: _cancellationTokenSource.Token);
		}
	}
```

Then create a Stop Scan function and use it only to manually stop a scan.  This will simply cancel the aforementioned `Cancellation Token` 

```
	/// <summary>
	/// A task as waiting for Stop request to complete is required
	/// </summary>
	/// <returns></returns>
	internal async Task ScanEnd()
	{
		if (_cancellationTokenSource?.IsCancellationRequested == false)
			await _cancellationTokenSource.CancelAsync();
	}
```

Now when the Scan is either manually stopped or timed out, it will cancel the inner `Cancellation Token` and appropriately set `IsScanning`

Never use `StartScanningForDevicesAsync` at all.   This is the issue with the API, it just doesn't make complete sense.