dotnet-bluetooth-le/dotnet-bluetooth-le #477: iOS Device disconnected while fetching services

MouratidisA, 2020-12-15:
I am working on a project with Xamarin.Forms for both Android and iOS platforms. I recently upgraded XCode to 12.3 and my iPad to iOS 14.3. 
My code for Connecting to a specified (using GUID) device and retrieving every characteristic update from a specified service (using GUID ) works fine in Android platforms but not for iOS (14.3).

When I try to execute ```await _targetDevice.GetServiceAsync(new Guid(deviceInfo.ServiceId));``` or ```await _backBraceDevice.GetServicesAsync();``` I get  "Device disconnected while fetching services" exception.

> Here is a sample code from my implementation with Plugin.BLE 2.1.1 that contains the Exception of the issue
```
    public class SamplingServiceTask
    {
        private IBluetoothLE _ble;
        private IAdapter _adapter;
        private ICharacteristic _deviceCharacteristic;
        private IService _deviceService;
        private IDevice _targetDevice;
 
        public SamplingServiceTask() { }

        public async Task RunSamplingTask(CancellationToken token)
        {
            Debug.WriteLine("Sampling Start");            

            await BluetoothHelpers.GetBluetoothState(Plugin.BLE.CrossBluetoothLE.Current);
            _ble = CrossBluetoothLE.Current;
            _adapter = CrossBluetoothLE.Current.Adapter;
            deviceInfo deviceInfo = null;

            deviceInfo = await deviceInfo.LoaddeviceInfo();
            if (deviceInfo == null)
            {
                Debug.WriteLine("Sampling - LoadDeviceInfo null ");               
                return;
            }

            if (!token.IsCancellationRequested)
            {
                token.ThrowIfCancellationRequested();

                var locProvider = DependencyService.Get<ILocationProvider>().IsLocationEnabled();
                 //If Location service is disabled 
                if (!locProvider)
                {                    
                    Debug.WriteLine("Location OFF");                  
                    return;
                }

                //If bluetooth is disabled 
                if (!_ble.IsOn)
                {
                    Debug.WriteLine("Bluetooth Adapter OFF");              
                    return;
                }

                //Connecting to device
                try
                {
                    if (!_adapter.IsScanning && _targetDevice == null)
                    {                       
                        _targetDevice = await _adapter.ConnectToKnownDeviceAsync(new Guid(deviceInfo.DeviceId), new ConnectParameters(false, true), token);
                    }
                }
                catch (Exception e)
                {
                    Debug.WriteLine("Connecting to device err: " + e.Message);                              
                }

                try
                {
                    if (_targetDevice != null)
                    {
                        //Discovering services
                        try
                        {                           
                            _deviceService = await _targetDevice.GetServiceAsync(new Guid(deviceInfo.ServiceId));
                        }
                        catch (Exception ex)
                        {
                            Debug.WriteLine("Discovering Service err: " + ex.Message);                         
                        }

                        if (_deviceService != null)
                        {
                            //Discovering characteristics                                
                            _deviceCharacteristic = await _deviceService.GetCharacteristicAsync(new Guid(deviceInfo.CharacteristicId));
                            if (_deviceCharacteristic != null)
                            {
                                // ... Register Notifications about characteristic....                                                      
                            }
                         
                        }
                        else
                        {
                            Debug.WriteLine("Service not Found");
                            await _ble.Adapter.DisconnectDeviceAsync(_targetDevice);                        
                        }

                    }
                    else
                    {
                        Debug.WriteLine("DNF - Device not Found");                       
                    }
                }
                catch (Exception e)
                {
                    Debug.WriteLine("Device Connection err: " + e);                   
                }
            }
            else
            {
                token.ThrowIfCancellationRequested();
            }

        }       
    }
```

## Steps to reproduce

1.  _targetDevice = await _adapter.ConnectToKnownDeviceAsync( DeviceGuid, new ConnectParameters(false, true), token);
2. _deviceService = await _targetDevice.GetServiceAsync( ServiceGuid);

## Expected behavior
The GetServiceAsync(ServiceGUID) should return the specified service form the Bluetooth device (works in Android Platforms).

## Actual behavior
The GetServiceAsync(ServiceGUID) throws an exception with message "Device disconnected while fetching services" (only in iOS platforms).

### Crashlog
{System.Exception: Device device disconnected while fetching services.
  at Plugin.BLE.Abstractions.Utils.TaskBuilder.FromEvent[TReturn,TEventHandler,TRejectHandler] (System.Action execute, System.Func`3[T1,T2,TResult] getCompleteHandler, System.Action`1[T] subscribeComplete, System.Action`1[T] unsubscribeComplete, System.Func`2[T,TResult] getRejectHandler, System.Action`1[T] subscribeReject, System.Action`1[T] unsubscribeReject, System.Threading.CancellationToken token) [0x000fd] in <12956cfcb4ff491aa0cc1f7ab64b7e72>:0 
  at Plugin.BLE.Abstractions.DeviceBase.GetServicesAsync (System.Threading.CancellationToken cancellationToken) [0x00050] in E:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\DeviceBase.cs:60 
  at BackBrace.Services.BraceSamplingServiceTask.RunSamplingTask (System.Threading.CancellationToken token) [0x0049e] in C:\BitBucket\backbrace-mobile\BackBrace\BackBrace\Services\SamplingServiceTask.cs:134 }

## Configuration

**Version of the Plugin:** V2.1.1

**Platform:** Both Android and iOS
- iOS 14.3
- Android 5.0 (API Level 21) till 10 (API Level 29)

**Device:** 
- iPad Pro (11-inch) (iOS 14.3)
- Xiaomi Redmi Note 5 (Android version: 9PKQ1 MIUI version: 11.0.3) 
- Xiaomi Pocophone F1 (Android version: 10qkq1 MIUI version: 12.0.2)

IgorStrekha, 2021-02-01:
the same problem. Any news?

RLXIWC, 2021-07-08:
Any news to this problem?

derekpapworth, 2021-08-02:
I'm having the same issue on trying to use this plugin for the first time.

is there any news on this and also, from the looks of this issue trail, is anyone on the project checking these issues as there doesn't seem any acknowledgment of this issue since first posted 7 months ago?

Thanks Derek

RLXIWC, 2021-08-02:
I have to reject my issue. My problem was that a cable on my PCB-Board caused a reset sometimes that's why fetching the service failed. My PCB also works with a Battery, if there is not much current left the Bluetooth-Connection fails due to the fact of the high current which BLE needs.

gerry123, 2024-11-20:
I'm having the same issue, Android works, iOS not. Any ideas?
Thanks Gerry

domneedham, 2025-02-04:
I hit this issue this morning, but found if I connect to the device, then scan it for services that works fine.

domneedham, 2025-02-04:
> > I hit this issue this morning, but found if I connect to the device, then scan it for services that works fine.
> 
> What version of .net do you use (I'm on "TargetFrameworks>net8.0-ios")?

I am using net9.0-ios and net9.0-android