dotnet-bluetooth-le/dotnet-bluetooth-le #480: _adapter.StartScanningForDevicesAsync() doesn't find any device with Android 11

Auto72, 2021-02-01:
## Steps to reproduce

1. Smartphone with Android 11
    Try to Scan for BLE devices

## Expected behavior
The BLE devices switched on are found.

## Actual behavior
It doesn't find any BLE device.

### Crashlog
none

## Configuration
Android 11

Version of the Plugin: 2.1.1 and 2.2.0-pre2

Platform: Android 11

Device: Samsung S10

Auto72, 2021-02-03:
The problem has been fixed.
It happened that I had the following "permissions" on the AndroidManifest.xml:

	<uses-feature android:name="android.hardware.location" android:required="false" />
	<uses-feature android:name="android.hardware.location.gps" android:required="false" />
	<uses-feature android:name="android.hardware.location.network" android:required="false" />
	<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />

Actually I don't know why :-) so I removed all of them and now it works also on Android 11.
With the above permissions inserted in the AndroidManifest.xml worked with previous Android version, like Android 10, 9, 8, 7, with no problem, but Android 11.

p-mobile, 2021-02-24:
I just tried sample code but some how it is only scanning few devices and all without names. 
I tried to compare by address and I don't see my iphone or PC in list. Iphone and pc are available if I scan through another application but not using this one.  I tried to run this app on my samsung galaxy s21. Not sure what I am missing. Please help.

Auto72, 2021-02-24:
It only scan BLE devices. 
Try to increase the ScanTimeout to 30 seconds and set the ScanMode to LowLatency.

adapter.ScanTimeout = 30000;
adapter.ScanMode = ScanMode.LowLatency;

And also ignore devices without name:

private void Adapter_DeviceDiscovered(object sender, Plugin.BLE.Abstractions.EventArgs.DeviceEventArgs e)
{
    IDevice device = e.Device;
    if (string.IsNullOrEmpty(device.Name)) { return; }
    
    // Do your stuff here
}

p-mobile, 2021-02-25:
Thanks for update. I tried suggested settings but I am still not getting any device list. As you mention it will scan only BLE devices, so if I want to get list of non BLE device as well then, what is the best way to achieve that?
Again thanks for your response.

closed 2024-05-06 by Auto72