dotnet-bluetooth-le/dotnet-bluetooth-le #483: DeviceConnectionLost not triggered when bluetooth is turned off [XF] [Android 10/11] [Plugin.Ble 2.1.1]

emanuel2811, 2021-02-26:
Hi guys, 

I've been using the nuget on Xamarin Forms and it's been a success, but I've run into a problem. If I connect to a device (pulsometer in this case) and then I turn off  bluetooth on the phone, no disconnection event is being fired. 

I'm certain this used to work at some point on my phone before I updated it to Android 11. But now I'm testing on another Android 10 device and it doesn't work either.

I also tried simulating pulsometers using nRF Connect but the behaviour is the same.

All that happened using phones with Android 10 & 11. I switched to another phone that has Android 9 and it works (the event is fired). Is any extra configuration required for Android 10/11?

My manifest has the following permissions:
```
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
```

Thanks in advance

## Steps to reproduce

1. Search for devices

2. Connect to pulsometer (Polar H10)

3. Turn off bluetooth on the phone (Android 10/11)

## Expected behavior
The DeviceConnectionLost event is triggered.

## Actual behavior
The event is not triggered and the device still appears in the ConnectedDevices list.

## Configuration

**Version of the Plugin:** 2.1.1

**Platform:** Android 10, Android 11

**Device:** Redmi Note 8, Samsung A51, Polar H10 7B6DF729 (pulsometer)

MKuckert, 2021-03-02:
I've seen the same behavior. Android simply doesn't close the connection. Our app detects the change in the adapters enabled state and closes the connection preactively if the adapter has been deactivated.

emanuel2811, 2021-03-02:
Thanks @MKuckert for the reply. The aproach you suggest solves my problem!

For what I've seen the issue is Android's fault and can be tackled with the workaround suggested above so I'm closing this issue.

closed 2021-03-02 by emanuel2811