dotnet-bluetooth-le/dotnet-bluetooth-le #486: iOs 14.4 Authentication is Insuficient [only iOS]

kvrscream, 2021-03-02:
Hi everyone.
After I update my iPhone 7 for iOS 14.4 my application on startUpdateAsync throw exception: Start Notifications: Error Error Domain=CBATTErrorDomain Code=5 "Authentication is insufficient." UserInfo={NSLocalizedDescription=Authentication is insufficient.}

My code: 
public async Task<List<TemperatureModel>> GetTemperatures(ICharacteristic characteristicTemperature, string deviceId)
        {

            List<TemperatureModel> medicoes = new List<TemperatureModel>();
            
            if (characteristicTemperature.CanUpdate)
            {
                characteristicTemperature.ValueUpdated += (o, args) =>
                {
                    byte[] bytes = args.Characteristic.Value;

                    Task.Delay(TimeSpan.FromSeconds(5)).Wait();

                    //Temperatura
                    if (characteristicTemperature.Uuid == "2a1c" ||
                        characteristicTemperature.Name.ToLower().Equals("temperature measurement") ||
                        characteristicTemperature.Uuid == "1809")
                    {

                        if (Buffer.ByteLength(bytes) > 0)
                        {
                            string hexadecimal = BitConverter.ToString(bytes);
                            string[] hexSplit = hexadecimal.Split('-');

                            //Temperatura
                            string medicaoHex = hexSplit[1] + hexSplit[0];
                            string result = int.Parse(medicaoHex, System.Globalization.NumberStyles.HexNumber).ToString();
                            string resultadoFormatado = result.Substring(0, 2) + "," + result.Substring(2, 2);
                            string temperatura = resultadoFormatado;

                            //Horário
                            string horaHex = hexSplit[4];
                            string minutoHex = hexSplit[3];
                            string segundoHex = hexSplit[2];

                            int hora = int.Parse(horaHex, System.Globalization.NumberStyles.HexNumber);
                            int minuto = int.Parse(minutoHex, System.Globalization.NumberStyles.HexNumber);
                            int segundo = int.Parse(segundoHex, System.Globalization.NumberStyles.HexNumber);
                            TimeSpan horario = new TimeSpan(hora, minuto, segundo);

                            
                            DateTime date = DateTime.Now;
                            DateTime updateDateTime = date.AddHours(-hora).AddMinutes(-minuto).AddSeconds(-segundo);

                            string userId = Application.Current.Properties["userId"].ToString();
                            medicoes.Add(
                                new TemperatureModel
                                {
                                    MeasurementDate = updateDateTime,
                                    Temperature = decimal.Parse(temperatura),
                                    DeviceId = deviceId,
                                    userId = userId
                                }
                            );
                            

                        }
                    }
                };

                await characteristicTemperature.StartUpdatesAsync();

            }

            return medicoes;

        }

## Steps to reproduce

1. Found bluetooth device

2. Connect to him

3. On startUpdateAsync see the exception

## Expected behavior
Pass on startUpdateAsync and return data values.

## Actual behavior
Catch exception on StartUpdateAsync 

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** 2.1.0
**Platform:** iOS 14.4

**Device:** iPhone 7

tompi, 2021-04-20:
@kvrscream Im guessing this is related to pairing/bonding ?

Can you connect and read the characteristic on your device using nrf connect(app from nordic)?

kvrscream, 2021-04-20:
Hi @tompi  thank's for help me.

This occurs when caracteristc.StartUpdatesAsync()

I try read and notify this caracteristic using nordic but the nordic app return this message:  "Error. Authentication is insufficient".

tompi, 2021-04-20:
@kvrscream  Then Im guessing there is something wrong with your device BLE implementation ? Can you read this characteristic on an android device using nrf connect?

NRF Connect is kind of my goto BLE implementation, if something is not working there, I blame the firmware/slave implementation ;-)

kvrscream, 2021-04-20:
@tompi  When I use Android I read and update characteristic normally. But in some devices Android show pair dialog every time when invoke caracteristc.StartUpdatesAsync().

closed 2021-05-29 by kvrscream

mert-akan, 2024-02-07:
Hi @kvrscream could you share a solution if you have solved this?

kvrscream, 2024-02-09:
Hi @mert-akan 
I solve update my info.plist.

Check if you have this permissions:

<key>NSBluetoothAlwaysUsageDescription</key>
	<string>Esse aplicativo precisa acessar o Bluetooth</string>
	<key>NSBluetoothPeripheralUsageDescription</key>
	<string>Esse aplicativo precisa acessar seu Bluetooth.</string>
	<key>NSLocationWhenInUseUsageDescription</key>
	<string>Esse aplicativo precisa acessar sua localização.</string>
	<key>UIBackgroundModes</key>