dotnet-bluetooth-le/dotnet-bluetooth-le #489: Is there a way to check current MTU?

Umer-Mahmood, 2021-03-09:
I am trying to send some 420 bytes of data to the connected ble device. BLE device is configured to set MTU to sometimes 20 bytes and sometimes 240 bytes. I am able to send all 420 bytes of data in chunks of 20 or 240 bytes smoothly. But I need to decide the chunk size based on what is the current value of negotiated MTU. 

## Steps to reproduce

1.

2.

3.

## Expected behavior
Should be able to check currently available MTU to chunk the data accordingly.
currentMTU = getcurrentMtu( );  

## Actual behavior
Unable to find a way to check current MTU

### Crashlog

## Configuration

IainS1986, 2021-03-16:
IDevice has a function `Task<int> RequestMtuAsync(int requestValue);`

This lets you request an MTU size and it will return the MTU available (might not be what you request).

iOS - you can't request an MTU size so this will just return the MTU available.

memsom, 2021-06-03:
Be aware that a *lot* of Android devices will fail to communicate and drop the connection if you attempt to set the MTU to anything other than 23. There is a bug in Samsung's BLE stack and it seems to have crept in to other manufacturers implementations too. With any Samsung phone that has a recent Android 10 or 11 update installed, you will be able to set an MTU, for example 512, and it will accept it and even respond with 512 as the new value. But after setting it, the controller with crash hard and drop the BLE connection. I guess there might be a value higher than 23 one might set, indeed [this](https://interrupt.memfault.com/blog/ble-throughput-primer) document seems to indicate that we should be able to set 185 or 512, but in my experience, setting the MTU almost never works. I know it used to. I have seen it work on Samsung phones prior to the January 2020 updates, and I have a Pixel 3a with 10 on where it still works too. But it really does not work (with this library at least) in practice.