dotnet-bluetooth-le/dotnet-bluetooth-le #490: Why doesn't it work

bbhxwl, 2021-03-10:
Why doesn't it work

android  

	<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  　<uses-permission android:name="android.permission.INTERNET" /></manifest>

 private void Adapter_DeviceDiscovered(object sender, Plugin.BLE.Abstractions.EventArgs.DeviceEventArgs e)
        {
            Console.WriteLine(e.Device.Name); 
        }

        private async  void Button_Clicked(object sender, EventArgs e)
        {
            var ble = CrossBluetoothLE.Current;
            var adapter = CrossBluetoothLE.Current.Adapter;

            adapter.DeviceDiscovered += Adapter_DeviceDiscovered;
            try
            {
                await adapter.StartScanningForDevicesAsync();

            }
            catch (Exception ex)
            {

            }
        }

janusw, 2021-04-10:
> Why doesn't it work

Hard to tell from the information you provide. Your code example is not complete and runnable, and you don't even say what exactly does not work. Probably you're not seeing any devices? `e.Device.Name` might be empty for many devices, or maybe there is an exception? (You don't have any diagnostics for this case.)

You also fail to specify which version of the plugin you use, on which platform (probably Android? version?) and on which device. Please provide more details ...

tompi, 2021-04-20:
My guess is its not working because he is not asking for permissions. ITs not enough to just declare your permissions on recent android versions, you need to actually ask for them runtime as well!

BTW, great to see some activity on this project @janusw ! There really arent many good options for working with BLE on xamarin, keep up the good work!

janusw, 2021-04-20:
> My guess is its not working because he is not asking for permissions. ITs not enough to just declare your permissions on recent android versions, you need to actually ask for them runtime as well!

Yeah, it's very probable that this is the problem. (It's mentioned in the ReadMe, though.)

Note that in the sample app we currently use [Plugin.Permissions](https://github.com/jamesmontemagno/PermissionsPlugin) to handle the runtime permissions, but we should probably switch to [Xamarin.Essentials](https://docs.microsoft.com/en-us/xamarin/essentials/permissions) soon.

> BTW, great to see some activity on this project @janusw ! There really arent many good options for working with BLE on xamarin, keep up the good work!

Thanks for the positive feedback. You are right that this project is pretty much the best options for BLE on Xamarin (most of the kudos for this go to @xabre and @smstuebe of course), but it was a bit neglected recently ...

rajnikantpatel23, 2021-04-23:
I am still not able to connect Android devices. Please help.

janusw, 2021-04-23:
> I am still not able to connect Android devices. Please help.

Are you actually requesting the location permission at runtime? (Btw, which Android version are you working with?)

If you don't, I recommend [Xamarin.Essentials.Permissions](https://docs.microsoft.com/de-de/xamarin/essentials/permissions) for that purpose. Essentially you need to check the location permission via:
```
Permissions.CheckStatusAsync<Permissions.LocationWhenInUse>()
```
and, in case it was not granted yet, request it via:
```
Permissions.RequestAsync<Permissions.LocationWhenInUse>()
```

Please let us know whether this fixes your problem.

rajnikantpatel23, 2021-04-23:
Yes, I am giving location permissions.

janusw, 2021-04-23:
> Yes, I am giving location permissions.

To be very clear here, note that giving permissions via

```
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```

(as you mentioned above) is not enough. For newer Android versions you also need to request permissions at runtime (e.g. via Xamarin.Essentials, as detailed in my last reply).

If you do all of that, I really have no idea why things are not working for you. You need to give us more information (at the very least: which version of Plugin.BLE are you using, which Android version do you try this on, etc).

Are you able to discover any devices with the `BLE.Client.Droid` sample application that is in the Plugin.BLE repo?

rajnikantpatel23, 2021-04-25:
I'm able to discover devices using Android app but when try to connect any device then getting 'gattcallback error: 133'.
For more detail, I'm trying to connect ESP32 Board with my Android Mobile(Android 10). Everything is working fine in iOS.

rajnikantpatel23, 2021-04-25:
I'm able to discover devices using Android app but when try to connect any device then getting 'gattcallback error: 133'.
For more detail, I'm trying to connect ESP32 Board with my Android Mobile(Android 10). Everything is working fine in iOS. (If needed I can attach sample code as well)

rajnikantpatel23, 2021-04-28:
@janusw Thanks for your help, 
I have resolved my Android issue using below code when connecting any device.

var parameters = new ConnectParameters(forceBleTransport: true);
await _adapter.ConnectToDeviceAsync(deviceToConnect, parameters, cancellationToken.Token);

Hoidi, 2021-08-12:
Close this issue?

rajnikantpatel23, 2021-08-12:
Issue is closed for me.

closed 2022-03-14 by janusw