dotnet-bluetooth-le/dotnet-bluetooth-le #494: System.InvalidCastException

theLahcim, 2021-03-16:
## Steps to reproduce

1. Plugin has been added as suggested.

2. Visual studio and Xamarin.Forms are updated to the latest version.

3. Code attached below

```c#
using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;
using Plugin.BLE.Abstractions.Exceptions;
using System;
using System.Collections.Generic;
using System.Collections.ObjectModel;
using System.Linq;
using System.Text;
using System.Threading.Tasks;

using Xamarin.Forms;
using Xamarin.Forms.Xaml;

namespace App1.Views
{
    [XamlCompilation(XamlCompilationOptions.Compile)]
    public partial class DeviceView : ContentPage
    {
        IBluetoothLE ble;
        IAdapter adapter;
        ObservableCollection<IDevice> deviceList;
        IDevice device;
        public DeviceView()
        {
            InitializeComponent();
            ble = CrossBluetoothLE.Current;
            adapter = CrossBluetoothLE.Current.Adapter;
            deviceList = new ObservableCollection<IDevice>();
            lv.ItemsSource = deviceList;
        }

        private void btnStatis_Clicked(object sender, EventArgs e)
        {
            var state = ble.State;
            this.DisplayAlert("Notice", state.ToString(), "OK !");
            if(state == BluetoothState.Off)
            {
                txtErrorBle.BackgroundColor = Color.Red;
                txtErrorBle.TextColor = Color.White;
                txtErrorBle.Text = "Your Bloutoth is off! Turn it on!";
            }
        }

        /// <summary>
        /// Scan the list of Devices
        /// </summary>
        /// <param name="sender"></param>
        /// <param name="e"></param>
        private async void btnScan_ClickedAsync(object sender, EventArgs e)
        {
            try { 
                deviceList.Clear();
                adapter.DeviceDiscovered += (s, a) =>
                {
                    deviceList.Add(a.Device); //Error
                };
                if (!ble.Adapter.IsScanning)
                {
                    await adapter.StartScanningForDevicesAsync();
                }
            }
            catch(Exception ex)
            {
                DisplayAlert("Notice", ex.Message.ToString(), "Error !");
            }

        }
}
```
4. Clicking on Scan Button

## Expected behavior
List of avaliable devices

## Actual behavior
System Exception

### Crashlog

System.InvalidCastException
  Message=Specified cast is not valid.

## Configuration

**Version of the Plugin:** 2.1.1

**Platform:** Compiled using Android 9.0 

**Device:** Xiami Mi A1

Scriptman, 2021-03-22:
I had the same issue. With my code. Roughly the same code as yours.

After disabling `lv.ItemsSource = deviceList;` the error went away. I think the ListView doesn't like the IDevice interface :)

closed 2021-03-24 by theLahcim

gnnsekhar, 2022-06-10:
Thanks Scriptman, for me this worked!

Edit2:
I was also missing the <ViewCell> in the DataTemplate of the ViewList. Which throws the same error.