dotnet-bluetooth-le/dotnet-bluetooth-le #497: BlueTooth Pairing Dialogue appers Twice

lrkPrecure, 2021-03-22:
have downloaded the Ble project and run it on Samsung galaxy s20+ against mobile debugging. Link of the project where we can see the issues https://github.com/xabre/xamarin-bluetooth-le Issues: "Bluetooth Pairing Dialogue appears Twice " ::::::::Not able to trace out where it is really occurring in second time, background "connect to device and pairing methods running on" I tried to do "Cancel" of "Bluetooth Pairing Dialogue" box, none of the intent.Action gets fired on "OnReceive" method. So it is difficult to fix when cancel button clicks. Please guide me ASAP. our project is almost on Prod.

## Steps to reproduce

1.Connect  and do pairing of the external device

2."Bluetooth Pairing Dialogue appears " :::::::twice continuously

3.  Not  able to see where it goes when we presses "Cancel" button

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:**  Android 10

**Device:**  Samsung galaxy s20+

sutterv, 2021-04-21:
I have the same issue. I have to connect (pair) a protected device. Is it possible to manage the paring programmaticaly ?

Auto72, 2023-05-26:
I have Google Pixel 7 phone and when I connect for the first time with the BLE device (not yet bonded) it asks to Pairing two times (pop up window), one after the other. One time should be enough. The BLE device requires secure communication and has no PIN and no keyboard.