dotnet-bluetooth-le/dotnet-bluetooth-le #502: Nothing Happened When Start Discovering (0 Device)

Boukri, 2021-04-11:
Hello everyone, When I press my button to scan nearly Bluetooth devices nothing happened
__________________________________________________________________________________________________
My code :
using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;
using System;
using System.Collections.ObjectModel;

public partial class DashboardPage : ContentPage
    {
        IBluetoothLE ble;
        IAdapter adapter;
        public  ObservableCollection<IDevice> Devices { get; set; }

        public DashboardPage()
        {
            InitializeComponent();
            ble = CrossBluetoothLE.Current;
            adapter = CrossBluetoothLE.Current.Adapter;
            Devices = new ObservableCollection<IDevice>();
        }

       
        private async void SearcheBtn_Clicked(object sender, EventArgs e)
        {
            try
            {
                var state = ble.State;
                if (state == BluetoothState.On)
                {
                    adapter.ScanTimeout = 30000;
                    Devices.Clear();
                    adapter.DeviceDiscovered += (s, a) =>
                    {
                        Devices.Add(a.Device);
                    };
                    if (!ble.Adapter.IsScanning)
                        await adapter.StartScanningForDevicesAsync();
                }
                else
                {
                    await Application.Current.MainPage.DisplayAlert("Attention", "Bluetooth is off", "Ok");
                }
            }
            catch (Exception ex)
            {
                await Application.Current.MainPage.DisplayAlert("Attention", "Error => "+ ex.Message, "Ok");
            }
            
        }
    }
___________________________________________________________________________________
Android permissions:
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  <uses-permission android:name="android.permission.BLUETOOTH" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

## Steps to reproduce

1. Search nearly devices

2. Connect to One specific device

3. Send Data (Text to print for ex)

## Expected behavior
I should get get a list of devices

## Actual behavior
Nothing happened

### Crashlog

No crashes 
## Configuration

Version of the Plugin => 2.1.1

Platform: Android 10.0 API 29

Device: Xiaomi M2007J20CG

Boukri, 2021-04-11:
Android permissions:
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  <uses-permission android:name="android.permission.BLUETOOTH" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

Boukri, 2021-04-12:
I figured that i should turn on my device location , I don't' why working with bluetooth requires location

janusw, 2021-04-12:
> I figured that i should turn on my device location , I don't' why working with bluetooth requires location

Well, it's simply required on Android. There is nothing we can do about it.

Note that this is mentioned in the README. So I'll just close this issue.

closed 2021-04-12 by janusw