dotnet-bluetooth-le/dotnet-bluetooth-le #503: Data Length Extension

philm001, 2021-04-11:
Hello all,

Just a quick question. I have a BLE application where I am sending 494 bytes through a notification.

After some testing, it appears that the program is unable to get the notification if it is above the 23 byte limit. Is there a way in the library to enable to data length extension so that I can get the 494 bytes?

janusw, 2021-04-12:
Hi @philm001 ,

> Just a quick question. I have a BLE application where I am sending 494 bytes through a notification.
>
> After some testing, it appears that the program is unable to get the notification if it is above [...]

I guess you are talking about Bluetooth 5 Advertisement Extensions, right? (These extended advertisement should be able to include up to 255 bytes in a single packet, and even more via chaining.)

Support for BT5 Advertisement Extensions has been added recently on Android (see #495 and #496). On iOS this is currently still not possible AFAIK.

We will soon prepare a release [2.1.2](https://github.com/xabre/xamarin-bluetooth-le/milestone/9), which will include this feature on Android.

philm001, 2021-04-12:
Ok great. Just to confirm, the package that is on the MS Nuget package manager, is that at the latest version with these fixes?

philm001, 2021-04-12:
Ok great, do you have a timeline for the release?

janusw, 2021-04-12:
> Ok great, do you have a timeline for the release?

I'd like to get it done within a week or so, but I'm not giving any guarantees ;)

In any case, I'm working on auto-building nuget packages via CI right now (cf. #501), which should make it much easier to try the master branch before the actual release.

If I read your explanations above correctly, then you only use xamarin-bluetooth-le for the receiver side of your application, while the transmitter side has a different implementation, right?

janusw, 2021-04-12:
> In any case, I'm working on auto-building nuget packages via CI right now (cf. #501), which should make it much easier to try the master branch before the actual release.

Done. The nuget packages for the current master branch are available here:
https://github.com/xabre/xamarin-bluetooth-le/suites/2478813162/artifacts/53361185

@philm001 Could you try those, in order to see if they work for you, and if they solve your problem?

philm001, 2021-04-12:
Hello Thank you for the update! I haven't installed a Nuget package manually before. According to my Nuget oackage manager, my Xamarin shared project is v2.1.1-24-g22bed21 which is the one that you gave me. The droid and iOS projects do not have this version.

I created another source that points to a local directory of the nuget packages. In my Manage package for Solution window,I clicked on the gear and added a directory that points to my local installation of the nuget packages. So I think that I am doing it correctly?

Like I said, the droid and iOS projects are unable to find the package even though the nuget manager settings are the same. As a result, when I run with the new update, the droid app is unable to grab the data (it also never gets notified that there is data).

Quick question, I thought that the data extension was released for BLE v4.2?

janusw, 2021-04-13:
> Hello Thank you for the update! I haven't installed a Nuget package manually before. According to my Nuget oackage manager, my Xamarin shared project is v2.1.1-24-g22bed21 which is the one that you gave me. The droid and iOS projects do not have this version.

You should make sure that all your projects reference the same version of the plugin (and yes, `v2.1.1-24-g22bed21` is the one from the GitHub Actions build mentioned above).

It might even be enough if only your shared project references the plugin (if your droid & iOS projects in turn reference the shared project).

> I created another source that points to a local directory of the nuget packages. In my Manage package for Solution window,I clicked on the gear and added a directory that points to my local installation of the nuget packages. So I think that I am doing it correctly?

Yes, sounds good. You need to set up an additional package source in VS, which consists of a local folder containing the nupkg files.

> Like I said, the droid and iOS projects are unable to find the package even though the nuget manager settings are the same. As a result, when I run with the new update, the droid app is unable to grab the data (it also never gets notified that there is data).

That's strange.

But yes, I also see that sometimes the local packages are not shown if I select "All sources". Seems one can work around that by selecting the specific local package source. (Note: I'm using VS for Mac.)
 
> Quick question, I thought that the data extension was released for BLE v4.2?

I'm afraid there was a misunderstanding on my side.

I somehow thought you were talking about BLE Advertising Extensions (with a payload of up to 255 instead of 31 bytes), which are actually a feature of BLE 5 (and the subject of #495).

However, you are obviously talking about BLE Data Length Extensions (DLE), which is a completely different thing, and I think it's true that those have been introduced in 4.2.

Sorry for the misunderstanding. In that case I'm not sure if the packages from our current master branch will help you, but you can still try.

janusw, 2021-04-14:
> Just a quick question. I have a BLE application where I am sending 494 bytes through a notification.
>
> After some testing, it appears that the program is unable to get the notification if it is above [...]

Could you tell us on which platform (operating system) you have tried this?

philm001, 2021-04-15:
For Visual studio I am on VS2019 windows 10

philm001, 2021-04-15:
Android 8.0 API 26

philm001, 2021-04-22:
Sorry for going dark on this. For now, we limited the transmission to 20 bytes and that is working fine. The application is able to work well with this data rate but am looking forward to the next release!

janusw, 2021-04-23:
> Sorry for going dark on this. For now, we limited the transmission to 20 bytes and that is working fine. The application is able to work well with this data rate but am looking forward to the next release!

Release 2.1.2 is out now, but I don't think it will help you with this. (You should still try it, though.)

Btw, I just found this ...
https://github.com/NordicSemiconductor/Android-BLE-Library/issues/42#issuecomment-436215164
... which claims that Android 6 and above should support DLE, and that there is no special API to enable it.

janusw, 2021-04-23:
Also here is an in-depth guide to DLE:
https://punchthrough.com/maximizing-ble-throughput-part-3-data-length-extension-dle-2/

I haven't read through it fully, but I hope it may be helpful to you.