dotnet-bluetooth-le/dotnet-bluetooth-le #514: Gatt write characteristic FAILED

diegoperfeito, 2021-05-06:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. I load a observable list with services list
2. ObservableCollection<servicesBT100> AuxservicosBT100 = new ObservableCollection<servicesBT100>();
foreach (IService servico in _servicos)
            {
                switch (servico.Id.ToString())
                {
                    case Constants.F1: // Função F1
                        var aux1 = new servicesBT100();
                        aux1.Servico = servico.Id.ToString();
                        aux1.Fn = "F1";
                        //aux1.Funcao = Application.Current.Resources["F1"].ToString ();
                        //aux1.DescFunc = Application.Current.Resources["DescF1"].ToString();
                        //aux1.DescFuncCompleta = Application.Current.Resources["DescF1Completa"].ToString();
                        aux1.Funcao = App.Resx.GetTraducao("F1");
                        aux1.DescFunc = App.Resx.GetTraducao("DescF1");
                        aux1.DescFuncCompleta = App.Resx.GetTraducao("DescF1Completa");

                        aux1.Farquivo = Application.Current.Resources["F1arquivo"].ToString();
                        aux1.IServico = servico;
                        AuxservicosBT100.Add(aux1);
                        break;

3. After Tapped Action 
4.var paginaFuncao3 = new TabbedPageF3(_characteristic, _numIntervalos, _charactIntervalos);
                    await Navigation.PushAsync(paginaFuncao3);
                    break;

5. The error happens when I send / submit the push feature to the next page. then the error Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: 'Gatt write characteristic FAILED.' happens.

6. Everything was working 60 days ago, something was updated / modified and the functionality started to give a problem only on Android. On iOS it's working fine.

### Crashlog

Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException
  Message=Gatt write characteristic FAILED.
  Source=mscorlib
  StackTrace:
  at Plugin.BLE.Android.Characteristic.InternalWrite (System.Byte[] data) [0x0004a] in D:\a\xamarin-bluetooth-le\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:110 
  at Plugin.BLE.Android.Characteristic+<>c__DisplayClass19_0.<WriteNativeAsync>b__0 () [0x00000] in D:\a\xamarin-bluetooth-le\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:81 
  at Plugin.BLE.Abstractions.Utils.TaskBuilder.FromEvent[TReturn,TEventHandler,TRejectHandler] (System.Action execute, System.Func`3[T1,T2,TResult] getCompleteHandler, System.Action`1[T] subscribeComplete, System.Action`1[T] unsubscribeComplete, System.Func`2[T,TResult] getRejectHandler, System.Action`1[T] subscribeReject, System.Action`1[T] unsubscribeReject, System.Threading.CancellationToken token) [0x000b3] in D:\a\xamarin-bluetooth-le\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\Utils\TaskBuilder.cs:48 
  at Plugin.BLE.Android.Characteristic.WriteNativeAsync (System.Byte[] data, Plugin.BLE.Abstractions.CharacteristicWriteType writeType) [0x000b2] in D:\a\xamarin-bluetooth-le\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:80 
  at Plugin.BLE.Abstractions.CharacteristicBase.WriteAsync (System.Byte[] data, System.Threading.CancellationToken cancellationToken) [0x00061] in D:\a\xamarin-bluetooth-le\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\CharacteristicBase.cs:92 
  at BT110MC.TabbedPageF3.picIntervalo_SelectedIndexChanged (System.Object sender, System.EventArgs e) [0x000b5] in E:\app\BT110MC\BT110\BT110\TabbedPageF3.xaml.cs:312 
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__7_0 (System.Object state) [0x00000] in /Users/builder/jenkins/workspace/archive-mono/2020-02/android/release/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/AsyncMethodBuilder.cs:1021 
  at Android.App.SyncContext+<>c__DisplayClass2_0.<Post>b__0 () [0x00000] in /Users/builder/azdo/_work/278/s/xamarin-android/src/Mono.Android/Android.App/SyncContext.cs:36 
  at Java.Lang.Thread+RunnableImplementor.Run () [0x00008] in /Users/builder/azdo/_work/278/s/xamarin-android/src/Mono.Android/Java.Lang/Thread.cs:36 
  at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00008] in /Users/builder/azdo/_work/278/s/xamarin-android/src/Mono.Android/obj/Release/monoandroid10/android-29/mcw/Java.Lang.IRunnable.cs:84 
  at (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.1(intptr,intptr)

## Configuration

**Version of the Plugin: Nuget 2.1.2

**Platform: Android 11 / Android 7

**Device: Xiaomi A3 / Moto G4 Plus

MKuckert, 2021-05-18:
Looks like you're writing from a different thread than the main thread.

Be aware of what's stated in the documentation:
> Characteristic/Descriptor Write: make sure you call characteristic.WriteAsync(...) from the main thread, failing to do so will most probably result in a GattWriteError.

memsom, 2021-06-03:
If you are using Xamarin.Essentials, wrap the write up like this:

```csharp
var success = false;
if (MainThread.IsMainThread)
{
   success  = await writer.WriteAsync(buffer);
}
else 
{
   success  = await MainThread.InvokeOnMainThreadAsync(async () =>  await writer.WriteAsync(buffer));
}
```

Something like that. But Android is not very good at writing a lot of data to characteristics. I implemented Reliable Writes in my fork of this library because of this. The Reliable Writes lets you write all the data in a transaction and then commit it, so that the controller can send all of the data reliably in one or more transmission windows.

Hackmodford, 2022-05-10:
I just ran into this issue on my app and have a solution.

1. Use Xamarin.Essentials `MainThread` to ensure every `WriteAsync` happens on the main thread.
```C#
_mainThread.InvokeOnMainThreadAsync(async () => characteristic.WriteAsync(data))
```
2. Use Gentlee's [SerialQueue](https://github.com/Gentlee/SerialQueue) to ensure all of my writes happen synchronously.

It seems that ensuring the Android implementation uses the main thread and performs writes in a serial queue would be a good addition to this library. Is there any interest in adding this to the library?