dotnet-bluetooth-le/dotnet-bluetooth-le #515: Request: Scanning filtering by manufacturer ID

kealist, 2021-05-07:
This is a feature request, but we recently had to abandon our use of this library due to not being able to scan filter by manufacturer ID (at least on Android, don't know what iOS world looks like).  I had added the functionality locally to the library just by passing through a manufacturerID and calling if that value was provided

```
                ScanFilter.Builder builder = new ScanFilter.Builder();
                builder.SetManufacturerData(manufacturerId.Value, Array.Empty<byte>() );
                ScanFilter filter = builder.Build();
                scanFilters.Add(filter);
```

kealist, 2021-05-07:
Sorry about not using template, but none of the template seems relevant to feature requests

janusw, 2021-05-07:
Thanks for the suggestion. That's a reasonable request.

Apart from filtering on Manufacturer data, it would also be good to have a way to filter on Service data. Right now one can only filter on Service UUID (by passing it directly to `IAdapter.StartScanningForDevicesAsync`).

Given all those different filtering options (cf. https://developer.android.com/reference/android/bluetooth/le/ScanFilter.Builder), it would be good to introduce more advanced filtering mechanisms to `IAdapter`.

And yes, one should also check the iOS APIs, in order to see what we can cover across platforms here.

igloo15, 2021-05-07:
Yeah this is a great feature to add since I have had difficult filtering on service ids and filtering on Manufacturer type did work for me on android.

janusw, 2022-03-14:
Scan filtering should even make it possible to scan while the screen is off on Android, see #565.

AdamDiament, 2022-03-24:
> Scan filtering should even make it possible to scan while the screen is off on Android, see #565.

@janusw I have a fork with working additional scan filtering (by MAC address for my needs). I can confirm this does let me scan whilst the screen is off on Android. Are you accepting PRs or should I hold off? 

As far as I can see, the only scan filter supported by iOS is by service UUID (see [the docs](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipherals)) If implemented, I  think it's worth making it clear that these additional filters will only have any effect on Android.

As a side note - most of the beacons I work with do not advertise a service UUID at all, annoyingly - hence me needing to go by MAC address for my app. It is going to make background scanning on iOS (which requires a filter) nigh on impossible for my needs. Any thoughts on that are welcome!

janusw, 2022-03-24:
> @janusw I have a fork with working additional scan filtering (by MAC address for my needs). I can confirm this does let me scan whilst the screen is off on Android. Are you accepting PRs or should I [...]

A PR would be more than welcome 👍 

> As far as I can see, the only scan filter supported by iOS is by service UUID (see [the docs](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipherals)). [...]

Agreed.

> As a side note - most of the beacons I work with do not advertise a service UUID at all, annoyingly - hence me needing to go by MAC address for my app. It is going to make background scanning on iOS (which also requires a filter) nigh on impossible for my needs. Any thoughts on that are welcome!

It might just be impossible to do background scanning on iOS. We have to live with the limits imposed by the OS and the API.

AdamDiament, 2022-03-24:
@janusw ok thanks for the reply, I'll have a look at getting a PR done. Are you ok if I update the android compilation to Android 12, or want me to leave it at 8 where it is?

janusw, 2022-03-24:
> @janusw ok thanks for the reply, I'll have a look at getting a PR done.

Great!

> Are you ok if I update the android compilation to Android 12, or want me to leave it at 8 where it is?

Don't have a strong opinion on that. I think up to now this library mostly took the conservation approach, only bumping the target version when it's really required. I think there were no significant bluetooth API changes in the latest Android versions (except for permissions), so we might as well stick to the Android 8 target for now, as long as there is no real advantage in updating.

AdamDiament, 2022-03-27:
@janusw @janusw I am trying to compile the BLE.Client.Droid project to get started on this PR, but I am getting a confusing build error.
```
    Error CS0006: Metadata file 'C:\projects\xamarin-bluetooth-le\Source\MvvmCross.Plugins.BLE.Droid\bin\Debug\MvvmCross.Plugins.BLE.Droid.dll' could not be found
```

This would seem to be a file that the build itself would create, so I'm not sure what's what. I've tried to usual suspects, deleting bin/obj folders, restarting, clean clone to a new folder etc. I'm running on windows with VS 2019 16.11.9

Have you encountered this at all?

AdamDiament, 2022-03-28:
I finally fixed this -  I opened a developer command prompt for vs 2019 to the ble.client.droid folder and ran MSbuild from there, and from the MVVMCross droid package. It built, then subsequent builds started working in visual studio. Ours is not to reason why...

AdamDiament, 2022-03-28:
@janusw here's the PR https://github.com/xabre/xamarin-bluetooth-le/pull/578

janusw, 2022-04-10:
Fixed via #578.

closed 2022-04-10 by janusw

janusw, 2022-05-01:
> Apart from filtering on Manufacturer data, it would also be good to have a way to filter on Service data. Right now one can only filter on Service UUID (by passing it directly to `IAdapter.StartScanningForDevicesAsync`).

Service-data filtering was added with #588.