dotnet-bluetooth-le/dotnet-bluetooth-le #516: How do I connect to multiple peripheral devices?

RK425, 2021-05-14:
## Expected behavior
I would like to connect to multiple BLE peripherals, I currently connect to a heart rate monitor and would like to also connect to a second or third device if they are found in the scan. Do I still just use the 
                __await adapter.ConnectToDeviceAsync(device);__ but provide the new IDevice? Or do I some how have to manage the multiple connections? Are there any examples available? 

**Version of the Plugin:** 2.1.1

**Platform:** Android 6 and greater

**Device:** Samsung S6

MKuckert, 2021-05-18:
I'd assume you'd be available to simply call `ConnecToDeviceAsync` for multiple devices 👍

RK425, 2021-05-19:
Thanks I will give it a go today.