dotnet-bluetooth-le/dotnet-bluetooth-le #522: Scanning only finds results on some devices and versions

TheDunco, 2021-07-01:
[Here](https://github.com/TheDunco/BLETest) is a link to the standalone GitHub repo where I have the code I am running.

## Steps to reproduce

1. Run the application linked above on an Android 10 device (preferably a Samsung Tablet) using Visual Studio 2019

2. Look at the debug output

## Expected behavior
The scan should result in the DeviceDiscovered callback being called and the message "Device Discovered!!!!!" followed by the name of the device printed out on the debug output window. I have a specific device I am trying to connect to or at least discover.

## Actual behavior
The scanner keeps rescanning and rescanning and never actually finds a single device. 

### Crashlog

Runtime error, no crashes.

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:** Android 10

**Device:** Samsung SM - T830 tablet

## Further information
I have tried this app on my phone, an LG V30 running Android 9 and it works perfectly. However, when I attempt to use it on the Samsung tablet it doesn't work at all. I've looked extensively into this issue on StackOverflow and this GitHub's issues but haven't found this exact situation or a solution anywhere. My AndroidManifest.xml includes all of the required permissions for Android 10 bluetooth...

```
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionCode="1" android:versionName="1.0" package="com.companyname.fca_grenlite">
  <uses-sdk android:minSdkVersion="29" android:targetSdkVersion="29"/>
  <application android:label="FCA_Grenlite.Android" android:theme="@style/MainTheme"></application>
  <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
  <uses-permission android:name="android.permission.BLUETOOTH" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
</manifest>
```
I set the min SDK version to 29 as well to make sure it was using the new Android 10 BACKGROUND_LOCATION permission.

This issue has been plaguing the project I'm working on for weeks and I've run out of options to try on my own -- I need help! If you require any more information please let me know and I will provide it asap.

Also note that the actual application is blank and everything I'm trying to do is taking place in the Visual Studio "Output" tab using the System.Diagnostics.Debug interface.

Thanks for any help you can provide,
Duncan

Auto72, 2021-07-21:
Try to remove this line from the manifest.

 <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"

I guess after it will work everywhere.

Auto72, 2021-07-22:
Also remember to have GPS sensor ON.

TheDunco, 2021-07-22:
I'll give it a try. I'm not currently able to work on this project but thanks for your feedback!