dotnet-bluetooth-le/dotnet-bluetooth-le #524: Bluetooth LE plugin does not support MVVMCross 8

BeardVis, 2021-07-10:
MvvmCross 8 switched from IMvxLog to Microsoft.Extensions.Logging. This is one of the reasons why the Bluetooth LE plugin cannot be used with MVVMCross 8.

frankfolsche, 2021-07-19:
I also ran in to this problem, I'm currently working on a PR that upgrades the plugin to MVVMCross 8. I hope to open this later today. 
A work around would be to use the non MVVMCross packages, then you will have to register de IBluetoothLE and the IAdapter yourself.

maciejpawlo, 2022-04-24:
Just ran into the same issue, MvvmCross 8.0.2, Plugin version 2.1.2

frankfolsche, 2022-06-18:
Unfortunately my [PR ](https://github.com/xabre/xamarin-bluetooth-le/pull/527) has not been approved yet.
For everyone running into this issue you can update your startup for Android and iOS with this code to prevent the broken plugin to load:
` public override IEnumerable<Assembly> GetPluginAssemblies()
        {
            return base.GetPluginAssemblies().Where(a => a.FullName != "MvvmCross.Plugins.BLE.iOS, Version=2.0.0.0, Culture=neutral, PublicKeyToken=null");
        }`
        then register the dependencies manually: 
        `
            Mvx.IoCProvider.LazyConstructAndRegisterSingleton<IBluetoothLE>(() => CrossBluetoothLE.Current);
            Mvx.IoCProvider.LazyConstructAndRegisterSingleton<IAdapter>(() => Mvx.IoCProvider.Resolve<IBluetoothLE>().Adapter);`