dotnet-bluetooth-le/dotnet-bluetooth-le #528: Disconnect Doesnt Complete After BT Connecting Successfully On The Second/X Try

JongChern, 2021-07-22:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Scan
2. Select Scanned Device
3. Power off said Bluetooth Device
4. Connect to said Device while it is powered off
5. Cancel Connect
5. Power on device, and connect
6. Connect is successful
7. Disconnect ~ Disconnect never completes

## Expected behavior
Disconnect should complete

## Actual behavior
Disconnect doesnt complete 

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
https://github.com/xabre/xamarin-bluetooth-le/tree/2.2.0-pre2

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
Windows 10

**Device:** e.g. HTC Sensation /i Phone 7 ...
Windows PC

janusw, 2022-12-10:
This is possibly a duplicate of #423. Does it still occur with 3.0.0-beta.2?

janusw, 2023-08-05:
Fixed via #721 in [3.0.0-beta.5](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.0.0-beta.5).

closed 2023-08-05 by janusw