dotnet-bluetooth-le/dotnet-bluetooth-le #530: Any plans on making this work for UWP?

dgerchcovich, 2021-07-26:
Awesome package.
But are there any plans on making this work for UWP?

Each time I try to connect to a device, or disconnect, or run device discovery, it fails.
Thanks,

adabru, 2023-10-08:
This is a duplicate of https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/issues/14.
@dgerchcovich I think you can close this one. I didn't try but https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/issues/14 suggests that UWP should be working from version 2.2.0 onwards.

janusw, 2023-10-09:
> This is a duplicate of #14.

True.

> I didn't try but #14 suggests that UWP should be working from version 2.2.0 onwards.

There has never actually been a final 2.2.0 release, but UWP support has been available since the 2.2.0-pre releases, and it is part of the stable release 3.0.0 now.

However, it may well be that individual features do not work fully yet. If you find any issues with the UWP support in 3.0.0, please open a new issue. I'm closing this one.

closed 2023-10-09 by janusw