dotnet-bluetooth-le/dotnet-bluetooth-le #535: App crash when WRITE descriptor value failed

ryanhz, 2021-09-01:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.   Keep constantly reseting Bluetooth module nrf52811
2.  App using BLE plugin to connect to it
3.  Then app will crash

## Expected behavior
Just throw error, no crash

## Actual behavior
App crash

### Crashlog

If something causes an exception paste full stack trace + Exception here

System.Exception: GATT: WRITE descriptor value failed
Descriptor.InternalWrite (System.Byte[] data)
Descriptor+<>c__DisplayClass8_0.<WriteNativeAsync>b__0 ()
TaskBuilder.FromEvent[TReturn,TEventHandler,TRejectHandler] (System.Action execute, System.Func`3[T1,T2,TResult] getCompleteHandler, System.Action`1[T] subscribeComplete, System.Action`1[T] unsubscribeComplete, System.Func`2[T,TResult] getRejectHandler, System.Action`1[T] subscribeReject, System.Action`1[T] unsubscribeReject, System.Threading.CancellationToken token)
Characteristic.StartUpdatesNativeAsync ()

## Configuration

**Version of the Plugin:** e.g. 2.1.2

**Platform:**  Android 11 

**Device:** Samsung Galaxy A31

MouratidisA, 2021-09-15:
Have you tested your nrf52811 implementation using other applications like [nRF Connect for Mobile](https://www.nordicsemi.com/Products/Development-tools/nrf-connect-for-mobile)?

TomasMalcanek, 2022-08-18:
This is also happening for me.
Same stacktrace, only Android from 10-13.
Wide amount of device models, mostly Galaxy devices, HUAWEI P30 Lite and P30 Pro.