dotnet-bluetooth-le/dotnet-bluetooth-le #536: Pre Release 2.2 Disconnect Issue in UWP

sisaacks, 2021-09-02:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Connect to a device

2. Then call _adapter.DisconnectDeviceAsync(connectedDevice);

3.

## Expected behavior
If I call await _adapter.DisconnectDeviceAsync(connectedDevice); I would expect the method to return and the device to be disconnected.  

## Actual behavior
There are two issues, if I call _await _adapter.DisconnectDeviceAsync(connectedDevice);_ the method never returns 
If I remove the await,  then the  device is removed from the _adapter.ConnectedDevices.  The device however is is the _adapater.DiscoveredDevices and its State is still **Connected**.  

### Crashlog

No crashes

## Configuration

**Version of the Plugin:** e.g. Pre Release 2.2.0

Platform is UWP on a windows box

janusw, 2022-12-10:
This is probably a duplicate of #423 and/or #528, right? Does it still occur with 3.0.0-beta.2?

janusw, 2023-08-05:
Fixed via #721 in [3.0.0-beta.5](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.0.0-beta.5).

closed 2023-08-05 by janusw