dotnet-bluetooth-le/dotnet-bluetooth-le #540: it doesn't work with android 9 and 10

andrettaadriano1, 2021-09-26:
Hi, I have developed an android only application with your library and it works perfectly with android 6 and 7, but on two devices with android 9 and 10 it does not work (does not detect any devices).
do i need to request additional permissions? 
Thanks

Karmageddonium, 2021-10-14:
Same thing on Android 10 using sample in the project

LeeRising, 2021-10-14:
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
Add this permission into manifest, and it must working

Karmageddonium, 2021-10-15:
> Add this permission into manifest, and it must working
"android.permission.ACCESS_BACKGROUND_LOCATION"

Also check if android settings location service is enabled (just in android settings as standart smartphone settings) or enable it in your code.
It appeared that it is not enough to request permission to location, it needs to be enabled as service in android.

ohgils, 2021-11-03:
> > Add this permission into manifest, and it must working
> > "android.permission.ACCESS_BACKGROUND_LOCATION"
> 
> Also check if android settings location service is enabled (just in android settings as standart smartphone settings) or enable it in your code. It appeared that it is not enough to request permission to location, it needs to be enabled as service in android.
> 
> Finally it worked for me

I have same problem. please give me information how to fix it.
ble.isOn is true, ble.State is BluetoothState.On, adapter.IsScanning is true.
but can't found devices. (iOS can found with same sourcecode)

This is my AndroidManifest.xml permisstions.
<uses-sdk android:minSdkVersion="18" android:targetSdkVersion="29" />
<application android:label="TDX_XG.Android" android:theme="@style/MainTheme"></application>
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
<uses-permission android:name="android.permission.INTERNET" />
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />

ohgils, 2021-11-03:
I have same problem. please give me information how to fix it.
ble.isOn is true, ble.State is BluetoothState.On, adapter.IsScanning is true.
but can't found devices. (iOS can found with same sourcecode)

This is my AndroidManifest.xml permisstions.
```
<uses-sdk android:minSdkVersion="18" android:targetSdkVersion="29" />
<application android:label="TDX_XG.Android" android:theme="@style/MainTheme"></application>
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
<uses-permission android:name="android.permission.INTERNET" />
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
```

ohgils, 2021-11-03:
> I have same problem. please give me some information how to fix it. ble.isOn is true, ble.State is BluetoothState.On, adapter.IsScanning is true. but can't found devices. (iOS can found with same sourcecode)
> 
> This is my AndroidManifest.xml permisstions.
> 
> ```
> <uses-sdk android:minSdkVersion="18" android:targetSdkVersion="29" />
> <application android:label="TDX_XG.Android" android:theme="@style/MainTheme"></application>
> <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
> <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
> <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
> <uses-permission android:name="android.permission.BLUETOOTH" />
> <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
> <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
> <uses-permission android:name="android.permission.INTERNET" />
> <uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
> ```

I found Samples BLE.Client.Droid. and it's no problem.
Thanks.

janusw, 2022-04-12:
> Add this permission into manifest, and it must working "android.permission.ACCESS_BACKGROUND_LOCATION"

Note that in the normal case ACCESS_FINE_LOCATION should be sufficient. ACCESS_BACKGROUND_LOCATION should only be required if you want to scan in a background service (on Android 10 and above), see https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#discover-local-devices.

janusw, 2022-04-12:
> Could anyone please explain why we need the location at all to scan for Bluetooth devices?

From https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#declare-android11-or-lower:

`ACCESS_FINE_LOCATION is necessary because, on Android 11 and lower, a Bluetooth scan could potentially be used to gather information about the location of the user.`

> And how to avoid using location completely..

Target Android 12. There you can use Bluetooth without the location permission, see https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#assert-never-for-location.

closed 2022-04-12 by janusw