dotnet-bluetooth-le/dotnet-bluetooth-le #553: Re - Scanning BLE devices

leomurmur, 2021-12-07:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.
When using the BLE plugin,
I have a problem when re-scanning BLE devices.
I created an object that scans BLE devices, finds the desired device and then reads the data from the characteristics.
An adapter is created in the object:
bluetoothAdapter = CrossBluetoothLE.Current.Adapter;
used:
       private void OnDeviceDiscovered (object sender, DeviceEventArgs args)
        {
            if (args.Device! = null &&! string.IsNullOrEmpty (args.Device.Name))
                _gattDevices.Add (args.Device);
        }
       await bluetoothAdapter.StartScanningForDevicesAsync ();
The scan is successful, the data is read too. After that, the object is destroyed.
Then the same object is created again to re-receive data, but in this case, scanning no longer works.
Could you please clarify why the second scan does not occur?
Thanks in advance.

## Steps to reproduce

1. Create an BLE Manager object, which provide scanning for device and read data 
     bluetoothAdapter = CrossBluetoothLE.Current.Adapter;
     private void OnDeviceDiscovered (object sender, DeviceEventArgs args)
        {
            if (args.Device! = null &&! string.IsNullOrEmpty (args.Device.Name))
                _gattDevices.Add (args.Device);
        }
       await bluetoothAdapter.StartScanningForDevicesAsync ();
      Read data;
2.  After that application destroy this object.

3. Application creates the same object to re-read the data, but scanning no longer works.

## Expected behavior
Tell us what should happen
Everything should work as in the first case.
## Actual behavior
Tell us what happens instead
Scanning no longer works.
### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

gluehbert, 2022-05-17:
Hello,

is there any solution for this issue? I can provide more information if needed.

Thank you

janusw, 2022-05-17:
> is there a solution to this problem?

Not yet.

> Have the same problem with the latest version 2.1.3, Android 10 & 11.

Thanks for the confirmation. So far it was not even clear on which platform it occurs.

gluehbert, 2022-05-18:
Currently there is a hotfix: You can disable Bluetooth and then enable it again. 

```
var bluetoothAdaper = ((BluetoothManager)Application.Context.GetSystemService(BluetoothService)).Adapter;
bluetoothAdapter.Disable();
while (bluetoothAdapter.State != State.Off)
    await Task.Delay(10);
bluetoothAdapter.Enable();
```

But that no longer works with Android 13 and any connected device loses the connection.

> [disable](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#disable())()
> This method was deprecated in API level Tiramisu. Starting with [Build.VERSION_CODES.TIRAMISU](https://developer.android.com/reference/android/os/Build.VERSION_CODES#TIRAMISU), applications are not allowed to enable/disable Bluetooth. Compatibility Note: For applications targeting [Build.VERSION_CODES.TIRAMISU](https://developer.android.com/reference/android/os/Build.VERSION_CODES#TIRAMISU) or above, this API will always fail and return false. If apps are targeting an older SDK ([Build.VERSION_CODES.S](https://developer.android.com/reference/android/os/Build.VERSION_CODES#S) or below), they can continue to use this API.