dotnet-bluetooth-le/dotnet-bluetooth-le #555: MonoTouchException with iOS and scanning for devices

cdavidyoung, 2021-12-17:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Download zip of this repo.

2. Build with Microsoft Visual Studio Community 2022 (64-bit) Version 17.0.2

3. Deploy to local device iPhone 12 Pro
4. Push the scan button

## Expected behavior
The app should find local bluetooth peripherals.  Indeed, I can cause it to work if I comment out the loop at line 316 in Adapter.cs of the Plugin.BLE.iOS project.  The line var cbuuid = array.GetItem<CBUUID>(i); is causing it to crash.  With this change much of the device information is not stored but otherwise the app works normally.

This site discusses the issue at length.  The user reported that the problem did not exist in the previous version of VS but started in version 17.0.2.

https://developercommunity.visualstudio.com/t/XamariniOS-getting-FoundationMonoTouch/1610258?viewtype=all

I expect that the problem is in VS but I would hope that a workaround could be found.  I have tried many ways of replacing the offending line but I can't seem to find a solution that compiles.

## Actual behavior
The app crashes with the indicated exception.

### Crashlog
Foundation.MonoTouchException: 'Objective-C exception thrown.  Name: NSRangeException Reason: *** -[__NSArrayM objectAtIndex:]: index 4479612000 beyond bounds [0 .. 0]
Native stack trace:
	0   CoreFoundation                      0x00000001819bd060 B2D21CFD-378C-36D5-BAF7-3F70599CFEFC + 626784
	1   libobjc.A.dylib                     0x000000019a031f54 objc_exception_throw + 60
	2   CoreFoundation                      0x0000000181ac6530 B2D21CFD-378C-36D5-BAF7-3F70599CFEFC + 1713456
	3   CoreFoundation                      0x0000000181953054 B2D21CFD-378C-36D5-BAF7-3F70599CFEFC + 192596
	4   Xamarin.PreBuilt.iOS                0x00000001042cdbec do_icall + 256
	5   Xamarin.PreBuilt.iOS                0x00000001042cc2c4 do_icall_wrapper + 296
	6   Xamarin.PreBuilt.iOS                0x00000001042c31b8 interp_exec_method_full + 2740
	7   Xamarin.PreBuilt.iOS                0x00000001042c1748 interp_runtime_invoke + 336
	8   Xamarin.PreBuilt.iOS                0x000000010412ee70 mono_jit_runtime_invoke + 900
	9   Xamarin.PreBuilt.iOS                0x00000001041db984 mono_runtime_invoke_checked + 148
	10  Xamarin.PreBuilt.iOS                0x00000001041dee08 mono_runtime_invoke + 144
	11  Xamarin.PreBuilt.iOS                0x00000001042ba0e0 xamarin_invoke_trampoline + 4584
	12  Xamarin.PreBuilt.iOS                0x00000001042bf9d4 xamarin_arch_trampoline + 148
	13  Xamarin.PreBuilt.iOS                0x00000001042c0548 xamarin_arm64_common_trampoline + 64
	14  CoreBluetooth                       0x000000019fbea8b0 9849C6D0-733E-3922-B0B6-691558DB9D03 + 71856
	15  CoreBluetooth                       0x000000019fbecabc 9849C6D0-733E-3922-B0B6-691558DB9D03 + 80572
	16  CoreBluetooth                       0x000000019fc26fc4 9849C6D0-733E-3922-B0B6-691558DB9D03 + 319428
	17  CoreBluetooth                       0x000000019fc1786c 9849C6D0-733E-3922-B0B6-691558DB9D03 + 256108
	18  libdispatch.dylib                   0x000000018162d914 A5CBAAB3-E389-3548-BAAC-FAB18411B94A + 10516
	19  libdispatch.dylib                   0x000000018162f660 A5CBAAB3-E389-3548-BAAC-FAB18411B94A + 18016
	20  libdispatch.dylib                   0x0000000181636de4 A5CBAAB3-E389-3548-BAAC-FAB18411B94A + 48612
	21  libdispatch.dylib                   0x000000018163798c A5CBAAB3-E389-3548-BAAC-FAB18411B94A + 51596
	22  libdispatch.dylib                   0x000000018163dab4 _dispatch_main_queue_callback_4CF + 772
	23  CoreFoundation                      0x0000000181975cd4 B2D21CFD-378C-36D5-BAF7-3F70599CFEFC + 335060
	24  CoreFoundation                      0x000000018192feac B2D21CFD-378C-36D5-BAF7-3F70599CFEFC + 48812
	25  CoreFoundation                      0x00000001819433b8 CFRunLoopRunSpecific + 600
	26  GraphicsServices                    0x000000019d2d338c GSEventRunModal + 164
	27  UIKitCore                           0x00000001842e36a8 8388EB03-002B-3B35-A78A-6A022894292E + 5346984
	28  UIKitCore                           0x00000001840627f4 UIApplicationMain + 2092
	29  Xamarin.PreBuilt.iOS                0x00000001042cdb90 do_icall + 164
	30  Xamarin.PreBuilt.iOS                0x00000001042cc2c4 do_icall_wrapper + 296
	31  Xamarin.PreBuilt.iOS                0x00000001042c31b8 interp_exec_method_full + 2740
	32  Xamarin.PreBuilt.iOS                0x00000001042c1748 interp_runtime_invoke + 336
	33  Xamarin.PreBuilt.iOS                0x000000010412ee70 mono_jit_runtime_invoke + 900
	34  Xamarin.PreBuilt.iOS                0x00000001041db984 mono_runtime_invoke_checked + 148
	35  Xamarin.PreBuilt.iOS                0x00000001041e2000 mono_runtime_try_invoke_array + 1272
	36  Xamarin.PreBuilt.iOS                0x000000010418b4f0 ves_icall_InternalInvoke + 1036
	37  Xamarin.PreBuilt.iOS                0x000000010419b780 ves_icall_InternalInvoke_raw + 108
	38  Xamarin.PreBuilt.iOS                0x00000001042cdb90 do_icall + 164
	39  Xamarin.PreBuilt.iOS                0x00000001042cc2c4 do_icall_wrapper + 296
	40  Xamarin.PreBuilt.iOS                0x00000001042c31b8 interp_exec_method_full + 2740
	41  Xamarin.PreBuilt.iOS                0x00000001042c1748 interp_runtime_invoke + 336
	42  Xamarin.PreBuilt.iOS                0x000000010412ee70 mono_jit_runtime_invoke + 900
	43  Xamarin.PreBuilt.iOS                0x00000001041db984 mono_runtime_invoke_checked + 148
	44  Xamarin.PreBuilt.iOS                0x00000001041e0bf8 mono_runtime_exec_main_checked + 128
	45  Xamarin.PreBuilt.iOS                0x0000000104113850 mono_jit_exec + 376
	46  Xamarin.PreBuilt.iOS                0x00000001042bf814 xamarin_main + 1924
	47  Xamarin.PreBuilt.iOS                0x00000001040db4cc main + 76
	48  dyld                                0x0000000104cb5a24 start + 520
'
Call stack:
>	[External Code]	

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
2.1.2

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
iOS 15.1.1

**Device:** e.g. HTC Sensation /i Phone 7 ...
iPhone 12 Pro

cdavidyoung, 2021-12-17:
BTW, I have also tried installing the Nuget version of this with my own Xamarin app and get the exact same exception.

dlchambers, 2022-01-17:
A month old and no comments... is it reasonable to expect that this will be addressed, or should I bail on Plugin.BLE and start looking for an alternative?

cdavidyoung, 2022-01-18:
If you use the source files and comment out the offending code it seems to work fine.  I am not even sure what that offending code does since it seems to be fully functional without it.  Let me know if you need any help using the repo as a starting point.

dlchambers, 2022-01-18:
Any chance that the offending code could be commented (or #ifdef'd) out & an updated official build be created that it available via NuGet?

cdavidyoung, 2022-01-18:
Well, they actually have a proposed fix.  I am skeptical though because they are not testing the fix in the environment in which the bug was discovered!

https://developercommunity.visualstudio.com/t/xamarinios-getting-foundationmonotouchexception-ob-1/1610258?from=email

janusw, 2022-03-14:
> https://developercommunity.visualstudio.com/t/xamarinios-getting-foundationmonotouchexception-ob-1/1610258?from=email

This thread sounds very strongly like it's a bug in VS2022 that does not affect VS2019. So I assume there is nothing to fix in xamarin-bluetooth-le? Can we close this issue?

dlchambers, 2022-03-14:
It's most definitely a bug in VS2022 (likely in Xamarin.iOS) which was not present in VS2019.
When I shifted back to using 2019 the bug went away.

I'd suggest leaving the bug open until it's confirmed fixed in VS2022. I realize that this forum is for the bluetooth plugin, but until MSFT fixes the issue the plugin is "broken" under 2022.

EddyCelis, 2022-04-27:
No problem on VS Enterprise 2022 version 17.1.4, so it was fixed.

janusw, 2022-04-27:
> No problem on VS Enterprise 2022 version 17.1.4, so it was fixed.

Awesome. Closing ...

closed 2022-04-27 by janusw