dotnet-bluetooth-le/dotnet-bluetooth-le #556: Question about how to get the advertisement ManufacturerData to display from BLE Device without connecting

SSmurf99, 2021-12-22:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

I am trying to display the ManufacturerData advertisement packet from the BLE Device without connecting to the device in the sample App but i am not sure how to..

## Expected behavior
Display the ManufacturerData advertisement packet preferably under each device in the mainscreen (scanning screen) without connecting to them.

## Actual behavior
N/A

### Crashlog

N/A

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:** e.g. iOS 14.2

**Device:** e.g. Iphone 8 Plus

janusw, 2022-03-14:
Try:

```
adapter.DeviceAdvertised += OnDeviceAdvertised;
await adapter.StartScanningForDevicesAsync();
```

and in your OnDeviceAdvertised handler have a look at the advertisement records:

```
        private async void OnDeviceAdvertised(object sender, DeviceEventArgs args)
        {
            var dev = args.Device;
            var adRecs = dev.AdvertisementRecords;
            foreach (var adrec in adRecs)
            {
                Debug.WriteLine(adrec);
            }
        }
```

janusw, 2022-04-12:
I hope that the above answers your question. I'll close this issue.

closed 2022-04-12 by janusw

densen2014, 2023-09-13:
in android and windows work fine, but in ios 16.6.1 AdvertisementRecords.Count always is 0, how can i fix this ?

Plugin.BLE: 3.0.0-beta.6
VS2022 MAUI Project

janusw, 2023-09-13:
> in android and windows work fine, but in ios 16.6.1 AdvertisementRecords.Count always is 0, how can i fix this ?

What works? It's unclear what you're doing and what you expect.

densen2014, 2023-09-13:
> > in android and windows work fine, but in ios 16.6.1 AdvertisementRecords.Count always is 0, how can i fix this ?
> 
> What works? It's unclear what you're doing and what you expect.

i am so sorry, it works fine,  just no return advertisementRecords with iBeacon `ManufacturerSpecificData`, only `ServiceData`.