dotnet-bluetooth-le/dotnet-bluetooth-le #557: Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED.

gadsaldanha, 2022-01-03:
I'm working on an app and I need to develop a feature whose goal is to CANCEL an ongoing operation.

But every time I write a new command and send it to the device, I get '**Gatt write characteristic FAILED**'.

Is it possible to cancel an ongoing operation?

janusw, 2022-04-14:
> But every time I write a new command and send it to the device, I get '**Gatt write characteristic FAILED**'.

You're not giving us much information to work with, but this error message tells me you're working on Android.

janusw, 2022-04-14:
> Is it possible to cancel an ongoing operation?

In principle yes. Most methods of `ICharacteristic` have a `cancellationToken` argument. This was even extended further recently, see #549. Is this sufficient for you, or do you need something else?