dotnet-bluetooth-le/dotnet-bluetooth-le #558: Is not working in Android 11

ClaudineiMonteiro, 2022-01-10:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

I developed an app to scan bluetooth devices with android 10, it's working normally, after updating to android 11 by google store demand, it stopped working. How do I resolve this?

## Steps to reproduce

1. I use this code:
2. if (bluetoothBLE.State == BluetoothState.Off)
            {
                await DisplayAlert("Caution", "Bluetooth disabled.", "OK");
            }
            else
            {
                list.Clear();
                foreach (var deviceConnected in adapter.ConnectedDevices)
                {
                    await adapter.DisconnectDeviceAsync(deviceConnected);
                }

                adapter.ScanTimeout = 10000;
                adapter.ScanMode = ScanMode.Balanced;

                adapter.DeviceDiscovered += (obj, a) =>
                {
                    try
                    {
                        if (!list.Contains(a.Device) && (!string.IsNullOrEmpty(a.Device.Name) && a.Device.Name.ToUpper().Contains("S3")))
                        {
                            list.Add(a.Device);
                        }
                    }
                    catch (Exception ex) 
                    {
                        App.MessageService.ShowAsync(ex.Message);
                    }
                };

                await adapter.StartScanningForDevicesAsync();
                if (list.Count == 0)
                {
                    await DisplayAlert("Attention", "No devices were found!", "OK"); 
                }
            }

2. But it doesn't find any device, if I go back to android 10 it works normally.

3.

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

janusw, 2022-03-14:
I cannot reproduce this. BLE.Client.Droid currently targets Android 10, but it still works correctly for me when I update it to Android 11. (You might run into trouble with Android 12, but Android 11 should work fine.)

Could you provide a reproducer that demonstrates the problem? Are you simply not finding any devices, or do you get any exception or error message?

janusw, 2022-03-15:
I just opened a PR that updates the BLE.Client.Droid sample from Android 10 to 11: #571

With this change, scanning still works well for me. Maybe you can try if that works for you?

janusw, 2022-04-07:
> My device is Sumsung Galaxy Note 20 Ultra and Android 12. I’m using the last version 2.1.2.

Alright, thanks for the information. That's pretty much the same setup that I have been testing as well: Targeting Android 11 and running on Android 12. Works perfectly for me. Lots of advertisements seen.

So, what I would suggest to you is:
1. Try running the sample app BLE.Client.Droid on the same device. Check if it sees any devices advertised.
2. Try running your app on another device (with another Android version).
3. And of course check that BT is turned on on your device and permissions are granted. Maybe try rebooting it (helped in #580).

If you do any of this, please report the results here. If you have questions, ask. If you fail to do any of this, we'll have to close this as "not reproducable", I'm afraid.

KittyDotNet, 2022-04-25:
not sure if its related, but im having an issue with devices not showing, but with a specific trigger. i have a host of new barcode scanners and tablets. if i take a new scanner and a new tablet, everything works perfectly start to finish. if i take that same scanner to aa second tablet, it works perfectly there too. if i take it back to the first one, it does not show up at all. not in DiscoveredDevces, or from DeviceConnected or DeviceAdvertsed, no where

janusw, 2022-04-26:
> getting it on both android 10 and 12

If you get it on Android 10 and 12, it's certainly not related to this issue, which says that something is not working on Android 11. Please open a new issue.

ClaudineiMonteiro, 2022-10-11:
Thanks, I’ll try.