dotnet-bluetooth-le/dotnet-bluetooth-le #559: Scan causes crash on iOS

dlchambers, 2022-01-13:
Scanning for devices on iOS causes a crash when scanning commences.
Works as expected on Android

## Steps to reproduce

1. Build `Samples > BLE.Client.iOS` and debug/deploy to iPhone.

2. On the **Scan For Devices** tab, tap **Scan**

3. Plugin.BLE crashes in `Task.Delay()` after calling `StartScanningForDevicesNativeAsync()` [near \Source\Plugin.BLE.Abstractions\AdapterBase.cs line 68]

## Expected behavior
Should scan & return discovered devices

## Actual behavior
Crashes immediately.
Crash cannot be handled by try/catch block.

### Crashlog
<pre>
Foundation.MonoTouchException
  Message=Objective-C exception thrown.  Name: NSRangeException Reason: *** -[__NSArrayM objectAtIndex:]: index 4387399264 beyond bounds [0 .. 0]
Native stack trace:
	0   CoreFoundation                      0x0000000181a59110 16FAA70C-278C-3561-859E-CEC407C2DC7C + 626960
	1   libobjc.A.dylib                     0x000000019a294d64 objc_exception_throw + 60
	2   CoreFoundation                      0x0000000181b62564 16FAA70C-278C-3561-859E-CEC407C2DC7C + 1713508
	3   CoreFoundation                      0x00000001819ef104 16FAA70C-278C-3561-859E-CEC407C2DC7C + 192772
	4   Xamarin.PreBuilt.iOS                0x0000000104c09c0c do_icall + 256
	5   Xamarin.PreBuilt.iOS                0x0000000104c082e4 do_icall_wrapper + 296
	6   Xamarin.PreBuilt.iOS                0x0000000104bff1d8 interp_exec_method_full + 2740
	7   Xamarin.PreBuilt.iOS                0x0000000104bfd768 interp_runtime_invoke + 336
	8   Xamarin.PreBuilt.iOS                0x0000000104a6ae70 mono_jit_runtime_invoke + 900
	9   Xamarin.PreBuilt.iOS                0x0000000104b17984 mono_runtime_invoke_checked + 148
	10  Xamarin.PreBuilt.iOS                0x0000000104b1ae08 mono_runtime_invoke + 144
	11  Xamarin.PreBuilt.iOS                0x0000000104bf60e0 xamarin_invoke_trampoline + 4584
	12  Xamarin.PreBuilt.iOS                0x0000000104bfb9f4 xamarin_arch_trampoline + 148
	13  Xamarin.PreBuilt.iOS                0x0000000104bfc568 xamarin_arm64_common_trampoline + 64
	14  CoreBluetooth                       0x000000019fc5afb4 6073DEB6-9809-361B-8EA0-2B9133ED8158 + 69556
	15  CoreBluetooth                       0x000000019fc5d02c 6073DEB6-9809-361B-8EA0-2B9133ED8158 + 77868
	16  CoreBluetooth                       0x000000019fc95960 6073DEB6-9809-361B-8EA0-2B9133ED8158 + 309600
	17  CoreBluetooth                       0x000000019fc86c2c 6073DEB6-9809-361B-8EA0-2B9133ED8158 + 248876
	18  libdispatch.dylib                   0x00000001816c9924 EDD169E1-D0DB-3808-A19E-99C1CD5A1C4C + 10532
	19  libdispatch.dylib                   0x00000001816cb670 EDD169E1-D0DB-3808-A19E-99C1CD5A1C4C + 18032
	20  libdispatch.dylib                   0x00000001816d2df4 EDD169E1-D0DB-3808-A19E-99C1CD5A1C4C + 48628
	21  libdispatch.dylib                   0x00000001816d399c EDD169E1-D0DB-3808-A19E-99C1CD5A1C4C + 51612
	22  libdispatch.dylib                   0x00000001816d9ac4 _dispatch_main_queue_callback_4CF + 772
	23  CoreFoundation                      0x0000000181a11d84 16FAA70C-278C-3561-859E-CEC407C2DC7C + 335236
	24  CoreFoundation                      0x00000001819cbf5c 16FAA70C-278C-3561-859E-CEC407C2DC7C + 48988
	25  CoreFoundation                      0x00000001819df468 CFRunLoopRunSpecific + 600
	26  GraphicsServices                    0x000000019d56b38c GSEventRunModal + 164
	27  UIKitCore                           0x0000000184381088 CD7F7BA2-A2C6-3727-AFF6-9BAAB60CC6AB + 5349512
	28  UIKitCore                           0x00000001840ff958 UIApplicationMain + 2092
	29  Xamarin.PreBuilt.iOS                0x0000000104c09bb0 do_icall + 164
	30  Xamarin.PreBuilt.iOS                0x0000000104c082e4 do_icall_wrapper + 296
	31  Xamarin.PreBuilt.iOS                0x0000000104bff1d8 interp_exec_method_full + 2740
	32  Xamarin.PreBuilt.iOS                0x0000000104bfd768 interp_runtime_invoke + 336
	33  Xamarin.PreBuilt.iOS                0x0000000104a6ae70 mono_jit_runtime_invoke + 900
	34  Xamarin.PreBuilt.iOS                0x0000000104b17984 mono_runtime_invoke_checked + 148
	35  Xamarin.PreBuilt.iOS                0x0000000104b1e000 mono_runtime_try_invoke_array + 1272
	36  Xamarin.PreBuilt.iOS                0x0000000104ac74f0 ves_icall_InternalInvoke + 1036
	37  Xamarin.PreBuilt.iOS                0x0000000104ad7780 ves_icall_InternalInvoke_raw + 108
	38  Xamarin.PreBuilt.iOS                0x0000000104c09bb0 do_icall + 164
	39  Xamarin.PreBuilt.iOS                0x0000000104c082e4 do_icall_wrapper + 296
	40  Xamarin.PreBuilt.iOS                0x0000000104bff1d8 interp_exec_method_full + 2740
	41  Xamarin.PreBuilt.iOS                0x0000000104bfd768 interp_runtime_invoke + 336
	42  Xamarin.PreBuilt.iOS                0x0000000104a6ae70 mono_jit_runtime_invoke + 900
	43  Xamarin.PreBuilt.iOS                0x0000000104b17984 mono_runtime_invoke_checked + 148
	44  Xamarin.PreBuilt.iOS                0x0000000104b1cbf8 mono_runtime_exec_main_checked + 128
	45  Xamarin.PreBuilt.iOS                0x0000000104a4f850 mono_jit_exec + 376
	46  Xamarin.PreBuilt.iOS                0x0000000104bfb834 xamarin_main + 1944
	47  Xamarin.PreBuilt.iOS                0x0000000104a174f4 main + 76
	48  dyld                                0x0000000105451aa4 start + 520

  Source=Xamarin.iOS
  StackTrace:
  at ObjCRuntime.Runtime.ThrowNSException (System.IntPtr ns_exception) [0x00000] in /Users/builder/azdo/_work/1/s/xamarin-macios/src/ObjCRuntime/Runtime.cs:449 
  at ObjCRuntime.Runtime.throw_ns_exception (System.IntPtr exc) [0x00000] in /Users/builder/azdo/_work/1/s/xamarin-macios/runtime/Delegates.generated.cs:392 
  at (wrapper native-to-managed) ObjCRuntime.Runtime.throw_ns_exception(intptr)
</pre>

## Configuration

**Version of the Plugin:** 2.1.2 (d/l from https://github.com/xabre/xamarin-bluetooth-le)

**Platform:** Windows 10 / Visual Studio 17.0.4 / Xamarin.Forms 5.0.0.2291 / Xamarin.iOS (on the Mac) 15.4.0.0 

**Device:** iPhone11 / iOS 15.2

dlchambers, 2022-01-17:
Might be a dup of #555

dlchambers, 2022-01-17:
Note: 
In Adapter.cs line ~315:
`array.Count()` is 2, so `array.GetItem<CBUUID>(0);` should be OK

EddyCelis, 2022-02-21:
Just installed Visual studio 2022 Professional 17.1.0 and it runs fine on IOS 15.
Hope Microsoft fixes this soon so that we can keep testing on out main development PC's (latest VS 2022).

janusw, 2022-03-14:
Runs fine here with VS for Mac 8.10.19 and Xamarin.iOS 15.6.0.3 (using an iPhone 8 with iOS 15.3.1). May be some sort of VS bug?

janusw, 2022-03-14:
> Just installed Visual studio 2022 Professional 17.1.0 and it runs fine on IOS 15.

So 17.0.4 is buggy, but 17.1.0 works? @dlchambers, can you confirm that?

dlchambers, 2022-03-14:
I will do some tests & post results.
Will take a week or so as I'm currently working on something else.

janusw, 2022-03-14:
Thanks! No hurry :)

(I would try it myself, but I currently don't have a working Windows dev environment, only Mac.)

EddyCelis, 2022-04-27:
No problem on VS Enterprise 2022 version 17.1.4

closed 2022-04-27 by janusw