dotnet-bluetooth-le/dotnet-bluetooth-le #560: WriteAsync in a Descriptor crash on iOS

Lykaiast, 2022-01-17:
Writting in a descriptor on iOS crash.
Works as expected on Android

## Steps to reproduce

1. Firstly connect to device and get service, characteristics and descriptor to use:
`await _bluetoothAdapter.ConnectToDeviceAsync(device, connectParameters); 
var service = await device.GetServiceAsync(Guid.Parse(BLUETOOTH_SERVICE_GUID)); sendCharacteristic = await service.GetCharacteristicAsync(Guid.Parse(BLUETOOTH_SEND_CHARACTERISTIC_GUID));
receiveCharacteristic = await service.GetCharacteristicAsync(Guid.Parse(BLUETOOTH_RECEIVE_CHARACTERISTIC_GUID));
IDescriptor descriptor = await receiveCharacteristic.GetDescriptorAsync(Guid.Parse(BLUETOOTH_RECEIVE_DESCRIPTOR_GUID));
byte[] activateNotifications = { 0x01, 0x00 };`

2. Secondly, invoke method to write in the descriptor (which it isn't null):
`await descriptor.WriteAsync(activateNotifications);`

3. Then crash and throw an exception

## Expected behavior
Write successfully in the descriptor, as it does in Android.

## Actual behavior
Throw an exception and crash the app.

### Crashlog

2022-01-17 10:38:01.736 ProjectTest.iOS[2782:1787708] *** Assertion failure in -[CBPeripheral writeValue:forDescriptor:], CBPeripheral.m:557

[0:] Foundation.MonoTouchException: Objective-C exception thrown.  Name: NSInternalInconsistencyException Reason: Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:
Native stack trace:
	0   CoreFoundation                      0x0000000182fe9d50 7A0C7B81-A5B6-36A6-B41C-C7C790076454 + 597328
	1   libobjc.A.dylib                     0x000000019a34e6a8 objc_exception_throw + 56
	2   Foundation                          0x00000001847ba99c 534D37B5-2B79-3993-AB18-822606B22ABF + 1190300
	3   CoreBluetooth                       0x00000001a55f413c 0F85FF82-39E0-30D2-ADDF-8691AB3EC9AA + 155964
	4   ProjectTest.iOS                      0x000000010584f560 ProjectTest.iOS + 21362016
	5   ProjectTest.iOS                     0x00000001057a1908 ProjectTest.iOS + 20650248
	6   ProjectTest.iOS                      0x00000001077ca32c ProjectTest.iOS + 54371116
	7    ProjectTest.iOS                0x000000010686cc3c ProjectTest.iOS + 38259772
	8    ProjectTest.iOS                0x0000000106876dcc  ProjectTest.iOS + 38301132
	9    ProjectTest.iOS                0x000000010682fd68 ProjectTest.iOS + 38010216
	10   ProjectTest.iOS               0x00000001077be744 ProjectTest.iOS + 54323012
	11   ProjectTest.iOS                0x000000010680c348 ProjectTest.iOS + 37864264
	12   ProjectTest.iOS                0x0000000104453f14 ProjectTest.iOS + 409364
	13   ProjectTest.iOS                0x0000000104620e7c ProjectTest.iOS + 2297468
	14   ProjectTest.iOS                0x00000001045a669c ProjectTest.iOS + 1795740
	15   ProjectTest.iOS                0x00000001045a649c ProjectTest.iOS + 1795228
	16   ProjectTest.iOS                0x0000000104620d38 ProjectTest.iOS  + 2297144
	17   ProjectTest.iOS                0x00000001045c37a8 ProjectTest.iOS + 1914792
	18   ProjectTest.iOS                0x0000000105772adc ProjectTest.iOS + 20458204
	19   ProjectTest.iOS                0x0000000104864b20 ProjectTest.iOS + 4672288
	20   ProjectTest.iOS               0x00000001078327b4 mono_pmip + 25300
	21   ProjectTest.iOS                0x00000001078df1dc mono_pmip + 732412
	22   ProjectTest.iOS               0x00000001078e2660 mono_pmip + 745856
	23   ProjectTest.iOS                0x00000001043f5418 ProjectTest.iOS + 21528
	24   ProjectTest.iOS                0x00000001043f7620 ProjectTest.iOS + 30240
	25  Foundation                          0x00000001846d2858 534D37B5-2B79-3993-AB18-822606B22ABF + 239704
	26  CoreFoundation                      0x000000018300a2b0 7A0C7B81-A5B6-36A6-B41C-C7C790076454 + 729776
	27  CoreFoundation                      0x000000018301a2d8 7A0C7B81-A5B6-36A6-B41C-C7C790076454 + 795352
	28  CoreFoundation                      0x0000000182f5d678 7A0C7B81-A5B6-36A6-B41C-C7C790076454 + 22136
	29  CoreFoundation                      0x0000000182f62aa8 7A0C7B81-A5B6-36A6-B41C-C7C790076454 + 43688
	30  CoreFoundation                      0x0000000182f75e1c CFRunLoopRunSpecific + 572
	31  GraphicsServices                    0x00000001a32d09a0 GSEventRunModal + 160
	32  UIKitCore                           0x00000001857a9600 D8869DEE-E238-3284-8C9A-623313659320 + 5162496
	33  UIKitCore                           0x000000018553eb08 UIApplicationMain + 2028
	34   ProjectTest.iOS                0x000000010584a0f8 ProjectTest.iOS + 21340408
	35   ProjectTest.iOS                0x00000001057aefc0 ProjectTest.iOS + 20705216
	36   ProjectTest.iOS                0x00000001044392a0 ProjectTest.iOS + 299680
	37   ProjectTest.iOS               0x0000000104864b20 ProjectTest.iOS + 4672288
	38   ProjectTest.iOS              0x00000001078327b4 mono_pmip + 25300
	39  ProjectTest.iOS               0x00000001078df1dc mono_pmip + 732412
	40   ProjectTest.iOS               0x00000001078e4450 mono_pmip + 753520
[... 44 lines cut ...]
	28  CoreFoundation                      0x0000000182f5d678 7A0C7B81-A5B6-36A6-B41C-C7C790076454 + 22136
	29  CoreFoundation                      0x0000000182f62aa8 7A0C7B81-A5B6-36A6-B41C-C7C790076454 + 43688
	30  CoreFoundation                      0x0000000182f75e1c CFRunLoopRunSpecific + 572
	31  GraphicsServices                    0x00000001a32d09a0 GSEventRunModal + 160
	32  UIKitCore                           0x00000001857a9600 D8869DEE-E238-3284-8C9A-623313659320 + 5162496
	33  UIKitCore                           0x000000018553eb08 UIApplicationMain + 2028
	34   ProjectTest.iOS                0x0000000101ece0f8 ProjectTest.iOS + 21340408
	35   ProjectTest.iOS                0x0000000101e32fc0 ProjectTest.iOS + 20705216
	36   ProjectTest.iOS                0x0000000100abd2a0 ProjectTest.iOS + 299680
	37   ProjectTest.iOS                0x0000000100ee8b20 ProjectTest.iOS + 4672288
	38   ProjectTest.iOS                0x0000000103eb67b4 mono_pmip + 25300
	39   ProjectTest.iOS                0x0000000103f631dc mono_pmip + 732412
	40   ProjectTest.iOS                0x0000000103f68450 mono_pmip + 753520
	41   ProjectTest.iOS               0x0000000103e9b0a8 ProjectTest.iOS + 54685864
	42   ProjectTest.iOS                0x000000010402fcc0 xamarin_localized_string_format_9 + 17612
	43   ProjectTest.iOS                0x0000000100ab9410 ProjectTest.iOS + 283664
	44  dyld                                0x0000000104558250 start + 444

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:** iOS 15.2

**Device:** iPhone 6s

GrantTem, 2022-01-31:
Maybe it helps：

1、Update ‘Microsoft Visual Studio Community 2022 (64-bit) - Preview’   to  ‘Version 17.1.0 Preview 4.0’

2、Install ‘Mobile development with C++’

Lykaiast, 2022-02-03:
I realize that on iOS is not necessary to write in a descriptor to unable notifications. That's whay it doesn't work

closed 2022-02-03 by Lykaiast