dotnet-bluetooth-le/dotnet-bluetooth-le #563: DeviceAdvertised event fired just one time for each device on iOS 15.2 / 15.3.1

XamMattia83, 2022-02-17:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Subscribe the DeviceAdvertised event

2. Perform  a "StartScanningForDevicesAsync"

3. Check / Debug the "DeviceAdvertised" event 

## Expected behavior
The DeviceAdvertised event should be triggered many times during a scan for each device. On Android side everything is working correctly.

## Actual behavior

The event is triggered one time for each device found during a scan. This happen only on iOS side.

### Crashlog

## Configuration

**Version of the Plugin:**  2.1.2

**Platform:**  iOS 15.3.1, 15.2 

**Device:** iPhone XR

janusw, 2022-04-12:
I can not reproduce this. In contrast to `DeviceDiscovered`, `DeviceAdvertised` should definitely be triggered more than once, and it does for me on an iPhone 8 with iOS 15. Is your `ScanTimeout` large enough for multiple advertisement to be seen?

XamMattia83, 2022-04-13:
Thanks for your answer! Yup, scan time is 60sec. On Android side everything is ok (DeviceAdvertised is triggered multiple times)  

I'm using the last version of Xamarin.Forms;  Here an extract of the code

PS: anyway your Lib is great ;)

        public DevicesScanPageViewModel(INavigationService navigationService, IDBLocalService dBLocalService, IAdapter BluetoohtAdapter, IBluetoothLE bluetoothLE, ITabSharedData tabSharedData) : base(navigationService)
        {
            CmdScanDispositivi = new Command(StartScanDispositivi);                                                 //init interfaccie (assegnaz gestore)
            CmdSyncDispositivi = new DelegateCommand(SincronizzaDispositiviRegistrati);
            CmdDeleteDevice = new DelegateCommand<BMBLUEModel.DeviceInfo>(EliminaDispositivo);

            _userDialogs = UserDialogs.Instance;
            _dbService = dBLocalService;                                                                            //interfaccia x database  
            _tabSharedData = tabSharedData;
            _bluetoothLE = bluetoothLE;
            _bluetoothAdapter = BluetoohtAdapter;                                                                   //adapter Bluetooth ed eventi relativi
            _bluetoothAdapter.DeviceDiscovered += OnDeviceDiscovered;
            _bluetoothAdapter.DeviceDisconnected += OnDeviceDisconnected;
            _bluetoothAdapter.DeviceConnectionLost += OnDeviceConnectionLost;
            _bluetoothAdapter.DeviceAdvertised += OnDeviceAdvertised;
        }

        private async void StartScanDispositivi()
        {
            New_BLE_Devices.Clear();

            if (!await PermissionsGrantedAsync())
            {
                await _userDialogs.AlertAsync(AppResources.PermessiScanNegati);
                return;
            }

            if (_bluetoothLE.IsOn == false)
            {
                //
                string str;

                if (_bluetoothLE.State == BluetoothState.Unknown)
                {
                    Debug.WriteLine(DateTime.Now + "BLE State sconosciuto");
                    _userDialogs.ShowLoading("Waiting BLE activation...", MaskType.Black);

                    while (_bluetoothLE.State == BluetoothState.Unknown)
                        await Task.Delay(100);

                    Debug.WriteLine(DateTime.Now + "BLE State OK");

                    _userDialogs.HideLoading();
                }
                else
                {
                    switch (_bluetoothLE.State)
                    {
                        case BluetoothState.Unavailable:
                            str = "BLE is not available on this device.";
                            break;
                        case BluetoothState.Unauthorized:
                            str = "You are not allowed to use BLE.";
                            break;
                        case BluetoothState.TurningOn:
                            str = "BLE is warming up, please wait.";
                            break;
                        case BluetoothState.TurningOff:
                            str = "BLE is turning off. Turn it on!";
                            break;
                        case BluetoothState.Off:
                            str = "BLE is off. Turn it on!";
                            break;
                        default:
[... 49 lines cut ...]
                }*/

                var pos = DispositiviAbbinati.IndexOf(dev);

                Xamarin.Essentials.MainThread.BeginInvokeOnMainThread(() =>
                {
                    DispositiviAbbinati[pos].Rssi = e.Device.Rssi;
                    DispositiviAbbinati[pos].OnlineStatus = true;
                    DispositiviAbbinati[pos].UpdateConnData();                          //mi serve x segnalare alla UI del cambiamento di dati
                });
            }
            else if (newdev != null)
            {
                //Dispositivo è nella lista dei nuovi Device

                Debug.WriteLine(e.Device.Name + "Advertiser...: Rssi = " + e.Device.Rssi);

                var pos = New_BLE_Devices.IndexOf(newdev);

                Xamarin.Essentials.MainThread.BeginInvokeOnMainThread(() =>
                {
                    New_BLE_Devices[pos].Update(e.Device);
                });
            }
        }

janusw, 2022-04-14:
> Thanks for your answer! Yup, scan time is 60sec. On Android side everything is ok (DeviceAdvertised is triggered multiple times)

Ok then. Do you get any kind of error message or exception when running on iOS? Does it run through for 60 sec without a second advertisement? Or does it maybe crash somewhere upon the first advertisement you see?

> I'm using the last version of Xamarin.Forms; Here an extract of the code

I haven't looked fully at your code, but I would suggest you start a bit simpler (in order to check if you get multiple advertisements per device), then add further complexity later and see if it starts breaking somewhere.

For the start, maybe it's a good idea to only subscribe to `DeviceAdvertised` but not to `DeviceDiscovered` and remove all the unnecessary clutter code. Only do a simple debug printout whenever you get an advertisement.

> PS: anyway your Lib is great ;)

Thanks, but it's not "my" lib at all ;)
Most of that praise goes to @xabre and @smstuebe. I'm just a sort of a "drop-in" maintainer for the time being ...

XamMattia83, 2022-04-26:
Hello! 

> Ok then. Do you get any kind of error message or exception when running on iOS? Does it run through for 60 sec without a second advertisement? Or does it maybe crash somewhere upon the first advertisement you see?

Nope, i dont receive any exception or error messages...and yes, sec are enough to get a second advertisement

I've also installed the BLE client from this repository, subscribed the "DeviceAdvertised" event  (i had to increase the timeout also since was only 10sec) and tested it on my device  but i have the same....:the event is raised one time/each device /each scan.
In this case i see on the debug Console some warning about ignoring parsing with some advertising entries (see below) but i guess it doesn't matter...

" [TRACE] (MvvmCross.Logging.MvxLog) Parsing Advertisement: Ignoring Advertisement entry for key kCBAdvDataTimestamp, since we don't know how to parse it yet.

[TRACE] (MvvmCross.Logging.MvxLog) Parsing Advertisement: Ignoring Advertisement entry for key kCBAdvDataRxPrimaryPHY, since we don't know how to parse it yet.

 [TRACE] (MvvmCross.Logging.MvxLog) Parsing Advertisement: Ignoring Advertisement entry for key kCBAdvDataRxSecondaryPHY"

janusw, 2022-04-26:
> I've also installed the BLE client from this repository, subscribed the "DeviceAdvertised" event (i had to increase the timeout also since was only 10sec) and tested it on my device but i have the same....:the event is raised one time/each device /each scan.

That's pretty strange. But good that you can reproduce it with the sample client.

Actually the handling of `DeviceAdvertised` vs `DeviceDiscovered` should be rather solid. You can see how it's done in the `AdapterBase` class, in particular:

https://github.com/xabre/xamarin-bluetooth-le/blob/d873642e0858fe4a21e3e4d77f8d50a5735df4a7/Source/Plugin.BLE.Abstractions/AdapterBase.cs#L194-L207

The discovered devices are kept track of via a `ConcurrentDictionary` with the device UUID as the key. I don't think this mechanism could be the root cause of your problem, but you could try to debug it anyway ...

janusw, 2022-04-26:
Ok, I think I found the reason. You need to set the argument `allowDuplicatesKey` to `true` when calling `StartScanningForDevicesAsync`. Otherwise 'duplicate' advertisements will be filtered away on iOS:

https://github.com/xabre/xamarin-bluetooth-le/blob/d873642e0858fe4a21e3e4d77f8d50a5735df4a7/Source/Plugin.BLE.Abstractions/Contracts/IAdapter.cs#L74-L78

RTFM ;)

closed 2022-04-26 by janusw

XamMattia83, 2022-04-27:
Great! Thanks again for your support ;)