dotnet-bluetooth-le/dotnet-bluetooth-le #565: Unable to scan while the screen is off on Android

yasserjarouf, 2022-03-07:
With Android and while scanning, if the screen is off then the device is unable to detect or connect to devices unless the screen is lit "unlock is not necessary".

Tried to disable battery optimizations and it doesn't help. Any ideas?

janusw, 2022-03-14:
This simply seems to be a limitation of the Android Bluetooth stack. See e.g.:
https://stackoverflow.com/questions/48077690/ble-scan-is-not-working-when-screen-is-off-on-android-8-1-0

Note that a workaround seems to be using a ScanFilter. xamarin-bluetooth-le currently does not support this, but it was requested in #515.

janusw, 2022-04-08:
> Note that a workaround seems to be using a ScanFilter. xamarin-bluetooth-le currently does not support this, but it was requested in #515.

Actually this is not completely true. In fact you can already use a limited scan filtering in the current version 2.1.2, by passing a service UUID to `adapter.StartScanningForDevicesAsync`. @yasserjarouf, maybe you can try that to see if it helps?

janusw, 2022-04-12:
I have verified that the new scan-filter capabilities introduced in #578 actually make it possible to receive advertisements on Android when the screen is off. The rate at which advertisements are received is a bit lower, compared to the case where the the app is in foreground, and even lower compared to the case where the app is in the background but the screen is on (where it is already lower than in the foreground). I tried this on a Samsung Galaxy S21 with Android 12. I assume this should also work with the service-UUID filtering which is available already in 2.1.2 (but haven't tried that).

Therefore I think this issue can be closed: Scanning on Android is possible when the screen is off, one only needs to use a scan filter.

closed 2022-04-12 by janusw

janusw, 2022-05-01:
Further scan filter options were added in #588.