dotnet-bluetooth-le/dotnet-bluetooth-le #566: I'll never scan out any Bluetooth.

bbhxwl, 2022-03-10:
I'll never scan out any Bluetooth.

code
```
 private async   void Button_Clicked(object sender, EventArgs e)
        {

            var ble = CrossBluetoothLE.Current;
            var rs = ble.State;
            var adapter = CrossBluetoothLE.Current.Adapter;
            adapter.ScanTimeout = 10000;
            adapter.DeviceDiscovered += Adapter_DeviceDiscovered;
           await adapter.StartScanningForDevicesAsync();
            Console.WriteLine();
            
        }
```

```
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionCode="1" android:versionName="1.0" package="com.companyname.app3">
    <uses-sdk android:minSdkVersion="21" android:targetSdkVersion="30" />
    <application android:label="App3.Android" android:theme="@style/MainTheme"></application>
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  <uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  <uses-permission android:name="android.permission.BLUETOOTH" />
  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />

</manifest>

```

bbhxwl, 2022-03-10:
Resolved pending breakpoint at 'D:\Project\App3\App3\App3\MainPage.xaml.cs:84,1' to void App3.MainPage.Adapter_DeviceDiscovered (object sender, Plugin.BLE.Abstractions.EventArgs.DeviceEventArgs e) [0x0000e].

janusw, 2022-03-14:
You're not telling use precisely enough what you're doing, how we can reproduce it or in which configuration you're building your stuff, and quite frankly ...

> I'll never scan out any Bluetooth.

... my English language parser just fails here with a semantics exception. I could try to guess what you mean (and what you're doing), but I don't have the time for that. Please open a proper bug report, and you will get proper help.

closed 2022-03-14 by janusw

bbhxwl, 2022-03-15:
> You're not telling use precisely enough what you're doing, how we can reproduce it or in which configuration you're building your stuff, and quite frankly ...
>
>
> ... my English language parser just [...]

At present, I can't search for ble devices in Android and IOS. I don't know why. I've only succeeded in windows.