dotnet-bluetooth-le/dotnet-bluetooth-le #567: Scan causes crash on Android detected by AppCenter

sandros06, 2022-03-11:
Hi, 

I post here because I received on AppCenter of my application a crash error from Plugin.BLE, I don't know how and why this error is triggered but I wish with the Xamarin exception stack, you can help me

## Steps to reproduce

Nothing sorry

### Crashlog

```
Xamarin Exception Stack:
System.ArgumentException: length
  at System.Array.Copy (System.Array sourceArray, System.Int32 sourceIndex, System.Array destinationArray, System.Int32 destinationIndex, System.Int32 length) [0x000b4] in <5a88e350e0674dd7b8ddba06471cdfe2>:0
  at Plugin.BLE.Android.Device.ParseScanRecord (System.Byte[] scanRecord) [0x00066] in <17887b5227df4ea9ab9b317d496489c6>:0
  at Plugin.BLE.Android.Device..ctor (Plugin.BLE.Android.Adapter adapter, Android.Bluetooth.BluetoothDevice nativeDevice, Android.Bluetooth.BluetoothGatt gatt, System.Int32 rssi, System.Byte[] advertisementData) [0x00017] in <17887b5227df4ea9ab9b317d496489c6>:0
  at Plugin.BLE.Android.Adapter+Api21BleScanCallback.OnScanResult (Android.Bluetooth.LE.ScanCallbackType callbackType, Android.Bluetooth.LE.ScanResult result) [0x00026] in <17887b5227df4ea9ab9b317d496489c6>:0
  at Android.Bluetooth.LE.ScanCallback.n_OnScanResult_ILandroid_bluetooth_le_ScanResult_ (System.IntPtr jnienv, System.IntPtr native__this, System.Int32 native_callbackType, System.IntPtr native_result) [0x00011] in <950ad56dbf9942d1a0c82aaea5a2cf2a>:0
  at (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.49(intptr,intptr,int,intptr)

```
## Configuration

**Version of the Plugin:** 2.1.1
**Android:** 9
**Android Build:** 50.2.A.3.77
**Manufacturer:** Sony
**Model:** Xperia XA2 (H3113)

janusw, 2022-03-14:
> **Version of the Plugin:** 2.1.1

This is not the latest version. Can you try 2.1.2?

flaviusmrn, 2023-04-10:
I have the same issue with plugin version 2.1.2.

Version of the Plugin: 2.1.2
Android: 13
Manufacturer: Samsung
Model: Galaxy M52 5G

Stack trace:
```
	System.ArgumentException: length
	  at System.Array.Copy (System.Array sourceArray, System.Int32 sourceIndex, System.Array destinationArray, System.Int32 destinationIndex, System.Int32 length) [0x000b4] in <79253351d305475da37a5006801de24f>:0 
	  at Plugin.BLE.Android.Device.ParseScanRecord (System.Byte[] scanRecord) [0x00066] in <3ed41cbd360644fa9d5166cf3cbddf57>:0 
	  at Plugin.BLE.Android.Device..ctor (Plugin.BLE.Android.Adapter adapter, Android.Bluetooth.BluetoothDevice nativeDevice, Android.Bluetooth.BluetoothGatt gatt, System.Int32 rssi, System.Byte[] advertisementData) [0x00017] in <3ed41cbd360644fa9d5166cf3cbddf57>:0 
	  at Plugin.BLE.Android.Adapter+Api21BleScanCallback.OnScanResult (Android.Bluetooth.LE.ScanCallbackType callbackType, Android.Bluetooth.LE.ScanResult result) [0x00026] in <3ed41cbd360644fa9d5166cf3cbddf57>:0 
	  at Android.Bluetooth.LE.ScanCallback.n_OnScanResult_ILandroid_bluetooth_le_ScanResult_ (System.IntPtr jnienv, System.IntPtr native__this, System.Int32 native_callbackType, System.IntPtr native_result) [0x00011] in <9c9d22043f50470ab178aa568b87d815>:0 
	  at (wrapper dynamic-method) Android.Runtime.DynamicMethodNameCounter.26(intptr,intptr,int,intptr)
```

janusw, 2023-04-10:
By now the latest stable version is 2.1.3. Does it happen with this as well? And with 3.0.0-beta2?
Does it occur with a specific device (that possibly sends out malformed advertisements)?

TomasMalcanek, 2023-04-20:
The Device.ParseScanRecord(System.Byte[] scanRecord) method is throwing a System.ArgumentException: length exception. This issue seems to occur after the OnDeviceDiscovered callback is called.

## Steps to reproduce

1. Start the app on the affected device(s) (Honor 50 Lite - Android 11, Galaxy A53 5G - Android 13)
2. Wait for the OnDeviceDiscovered callback to be called.
3. Attempt to call the Device.ParseScanRecord method with a byte array as the scanRecord argument.

## Expected behavior
The Device.ParseScanRecord method should be able to parse the byte array without throwing an exception.

## Actual behavior
The Device.ParseScanRecord method is throwing a System.ArgumentException: length exception.

### Crashlog

```
Array.Copy (System.Array sourceArray, System.Int32 sourceIndex, System.Array destinationArray, System.Int32 destinationIndex, System.Int32 length)
Device.ParseScanRecord (System.Byte[] scanRecord)
Plugin.BLE.Android.Device..ctor (Plugin.BLE.Android.Adapter adapter, Android.Bluetooth.BluetoothDevice nativeDevice, Android.Bluetooth.BluetoothGatt gatt, System.Int32 rssi, System.Byte[] advertisementData) [0x00017] in <e84f194db96045279f39298a58a796ea>:0
Adapter+Api21BleScanCallback.OnScanResult (Android.Bluetooth.LE.ScanCallbackType callbackType, Android.Bluetooth.LE.ScanResult result)
ScanCallback.n_OnScanResult_ILandroid_bluetooth_le_ScanResult_ (System.IntPtr jnienv, System.IntPtr native__this, System.Int32 native_callbackType, System.IntPtr native_result)
```

## Configuration

**Version of the Plugin:** 2.1.3

**Platform:** Android 11 / Android 13

**Device:** Honor 50 Lite, Galaxy A53 5G

janusw, 2023-04-20:
> The Device.ParseScanRecord(System.Byte[] scanRecord) method is throwing a System.ArgumentException: length exception. This issue seems to occur after the OnDeviceDiscovered callback is called.

@TomasMalcanek Thanks for confirming this issue. Could you please verify if this occurs with version `3.0.0-beta.3` as well? (I assume it does, but I would like to make sure).

In the end, this is most probably a bug in a specific remote device that sends out malformed advertisements. But since it has been reported by several people already, maybe we should make the parsing code more robust against such issues?

@TomasMalcanek, @flaviusmrn  Could one of you possibly show us the complete (raw) advertising data for the case where you see the exception? From what kind of devices do these advertisements come?

TomasMalcanek, 2023-04-20:
@janusw I will upgrade to version `3.0.0-beta.3` to see if there is any difference.

I can try to provide advertising data, but we have a crash in AppCenter that started about 2 weeks ago. Since then there are only 2 usable reports. The expected release of the next version is about a month from now, so expect some more info from me later.

I think we could log the advertising data only in the `OnDeviceDiscovered` callback. How could we access it?

janusw, 2023-04-20:
> @janusw I will upgrade to version `3.0.0-beta.3` to see if there is any difference.

Thanks!

> I can try to provide advertising data, but we have a crash in AppCenter that started about 2 weeks ago. Since then there are only 2 usable reports. The expected release of the next version is about [...]

Well, using the Plugin.BLE nupkg, I'm afraid you can't.

You could debug using our sample client, setting a break point in the first lines of `Plugin.BLE.Android.Device.ParseScanRecord` and inspect the raw byte array. But that requires that you can reproduce the failure yourself (reliably). If it's only seen by your users (and reported to you via AppCenter), then this won't work.

Things that we could do to address this issue:
1. Verify that the length byte is correct (before copying data out of the byte array) and simply dispose malformed advertisements.
2. If the length byte is wrong, make sure to only copy data until the end of the byte array and return this (possibly cropped) advertisement.
3. Check the length byte and throw an exception if there is a problem (including a hex dump of the byte array?).

Which solution would you prefer? The first one is the most simple, but we don't get any information about malformed advertisements and cannot investigate further. The second one would at least return the advertisement data in some form, but it could create problems down the line (depending on the type of advertisement). The third one is probably the best option IMHO. If I implement that, would you use it?

TomasMalcanek, 2023-04-20:
> Well, using the Plugin.BLE nupkg, I'm afraid you can't.
> 
> You could debug using our sample client, setting a break point in the first lines of Plugin.BLE.Android.Device.ParseScanRecord and inspect the raw byte array.

I can fork the beta version and edit it to log the advertisement array. That way we will see it in the crash log next time.

> Which solution would you prefer? The first one is the most simple, but we don't get any information about malformed advertisements and cannot investigate further. The second one would at least [...]

The third solution should be the best option, but ideally, throw it at a place that can be catched and handled to prevent application hard crashes.

I will fork the version you suggest using.

janusw, 2023-04-20:
> The third solution should be the best option, but ideally, throw it at a place that can be catched and handled to prevent application hard crashes.

That was my idea, but yeah, since the parsing is done in the scan callbacks, having it catchable might be a problem ...

> I can fork the beta version and edit it to log the advertisement array. That way we will see it in the crash log next time.

That would be awesome. Please use the latest release (3.0.0-beta.3), if possible (if that is an issue for you, 2.1.3 would also be ok). And please report back your findings here ...

janusw, 2023-08-04:
This should be fixed via #719. Exceptions in `ParseScanRecord` are being caught now and incorrect records are disposed (with some debug output written to the log).

closed 2023-08-04 by janusw

janusw, 2023-08-05:
The fix is available in [3.0.0-beta.5](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.0.0-beta.5).