dotnet-bluetooth-le/dotnet-bluetooth-le #568: Android permission ACCESS_FINE_LOCATION

timdinhdotcom, 2022-03-13:
We had a hard time getting our app approved on the Google Play store because it request ACCESS_FINE_LOCATION permission.

Is it essential to have this permission for a bluetooth BLE app? 

## Expected behavior
Should not required ACCESS_FINE_LOCATION permission

## Actual behavior
N/A

### Crashlog

N/A

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:** Android 12

**Device:** Pixel 5

smstuebe, 2022-03-13:
You need to set the permissions you need for your use cases :) https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

janusw, 2022-03-14:
Actually there is a new bluetooth permission scheme on Android 12, and ACCESS_FINE_LOCATION is not strictly required any more!

In order to make use of this, [xamarin-bluetooth-le](https://github.com/xabre/xamarin-bluetooth-le) needs to be updated to support Android 12 and its new permission.

jamesmontemagno, 2022-03-20:
A quick work around is to just target API 31 for now and you will be good to go.

I am testing out an implementation as it is just permissions and I don't know if the actually library needs to be adjusted.

AndroidManifest.xml:

```xml
        <!-- FOR 30 AND BELOW -->
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />

         <!-- for 31 and above -->
	<uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
	<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

	<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
```

Source code to check permissions:

```csharp
public class PlatformHelpers
{
  public async Task<PermissionStatus> CheckAndRequestLocationPermission()
        {
            PermissionStatus status;
            if ((int)Build.VERSION.SdkInt >= 32)
            {
                status = await Permissions.CheckStatusAsync<BluetoothSPermission>();

                if (status == PermissionStatus.Granted)
                    return status;

                status = await Permissions.RequestAsync<BluetoothSPermission>();
            }
            else
            {
                status = await Permissions.CheckStatusAsync<Permissions.LocationWhenInUse>();

                if (status == PermissionStatus.Granted)
                    return status;

                if (status == PermissionStatus.Denied && DeviceInfo.Platform == DevicePlatform.iOS)
                {
                    // Prompt the user to turn on in settings
                    // On iOS once a permission has been denied it may not be requested again from the application
                    return status;
                }

                //if (Permissions.ShouldShowRationale<Permissions.LocationWhenInUse>())
                {
                    await MyApp.Current.MainPage.DisplayAlert("Permission required", "Location permission is required for bluetooth scanning. We do not store or use your location at all.", "OK");
                }

                status = await Permissions.RequestAsync<Permissions.LocationWhenInUse>();
            }
            return status;
        }
    }

    public class BluetoothSPermission : Xamarin.Essentials.Permissions.BasePlatformPermission
    {
        public override (string androidPermission, bool isRuntime)[] RequiredPermissions => new List<(string androidPermission, bool isRuntime)>
        {
            ("android.permission.BLUETOOTH_SCAN", true),
            ("android.permission.BLUETOOTH_CONNECT", true)
        }.ToArray();
    }
```

janusw, 2022-03-20:
> A quick work around is to just target API 31 for now and you will be good to go.
> 
> I am testing out an implementation as it is just permissions and I don't know if the actually library needs to be adjusted.

Yeah, you might be right. In principle it's just permissions. I was wondering if it's necessary for the library itself to target API 31 for this to work, but probably it's not. We should check.

In any case, we should at least make the following changes wrt the Android 12 permissions:
* update the ReadMe
* update the sample client to Android 12 and use the new permissions

jamesmontemagno, 2022-03-21:
This is my assumption at least... I need to test it out that is for sure.  I bought an Android 12 device recently so my plan is to test it out this week.

I don't believe that your code needs to change anything unless you are checking persmissions... but i just need to investigate.

janusw, 2022-04-06:
> Source code to check permissions:

Thanks for sharing this snippet, @jamesmontemagno. But ideally this runtime handling of Bluetooth permissions should be integrated into Xamarin.Essentials, no?

closed 2022-04-11 by janusw

sisaacks, 2022-04-26:
Hello....@janusw and @jamesmontemagno I have recently upgraded to Android 12.  I added the permissions to my Manifest but I am still getting an error when trying to scan for BT that I need Bluetooth Scan.  But I have added it to my manifest....I am also targeting API 31.  I am not sure what I need to do to fix this???

janusw, 2022-04-26:
> Hello....@janusw and @jamesmontemagno I have recently upgraded to Android 12. I added the permissions to my Manifest but I am still getting an error when trying to scan for BT that I need Bluetooth Scan.

Adding the permissions to the manifest is not enough. You also need to add runtime handling of the permissions. Have a look at what I did in #582, especially 5b77f8e0f89a6810767aad038aa1ffafb7da02e2.

sisaacks, 2022-04-26:
> Adding the permissions to the manifest is not enough. You also need to add runtime handling of the permissions. Have a look at what I did in #582, especially [...]

I went to your example, but I still have some questions.  I implemented the  class in the MainActivity and it works...however I still need LocationPermissions to pick up devices when scanning.  If I dont ask for that permission the devices are not picked up by the BLE scan

beeradmoore, 2022-09-03:
@jamesmontemagno, I think this is meant to be >= 31. Android 12 is 31 and 12L is 32. If that is correct could you please update that to stop people copy/pasting the incorrect lines because who reads samples properly :P 

> ```cs
>             PermissionStatus status;
>             if ((int)Build.VERSION.SdkInt >= 32)
>             {
> ```

PR #582 (2022-04-11): BLE.Client: update to Android12
This updates the target framework version to 12 for BLE.Client.Droid, uses the new Bluetooth permission scheme and adds the necessary runtime handling for the new Bluetooth permissions (while keeping backward compatibility to older Android versions). I tested it on Android 11 and 12, and verified that the sample app sees BLE advertisements on both versions.

It is proof that the library works well with Android 12 (without any changes necessary) and only proper permission handling is needed in client apps. It also provide a blueprint of how to handle the new perm [... cut]