dotnet-bluetooth-le/dotnet-bluetooth-le #569: No BLE disocvery during scan in Service mode when screen is off

yeohcl99, 2022-03-14:
adapter.DeviceDiscovered += (s, a) =>{} works well when the screen is on. But when screen is off on Android 9 the  adapter.DeviceDiscovered += (s, a) =>{} no longer able to discover any near by BLE in background service mode. However when the screen comes on immediately able to scan the surrounding ble. Any idea how to resolve this issue?

closed 2022-03-14 by yeohcl99

AdamDiament, 2022-03-23:
@yeohcl99 how did you resolve this issue?

yeohcl99, 2022-03-24:
If i read the link correctly.. there is no way to scan when the screen is off or off focus. I am looking at other package see if anyone has able to do it...

janusw, 2022-04-12:
> If i read the link correctly.. there is no way to scan when the screen is off or off focus.

Not true. It *is* possible (on Android) if you use a scan filter.

yeohcl99, 2022-04-13:
Could you please help to provide the sample code how to use the scan filter so the ble can scan when device is screen off or de focus? Thanks.

janusw, 2022-04-13:
> I tried the following but error on ScanFilterOptions()
> 
> ```
>      var scanFilterOptions = new ScanFilterOptions();
>         scanFilterOptions.ServiceUuids = new[] { "", ""}; // cross platform filter
>         scanFilterOptions.ManufacturerIds = new[] { 1, 2, 3}; // android only filter
>         scanFilterOptions.DeviceAddresses = new[] { "80:6F:B0:43:8D:3B", "80:6F:B0:25:C3:15"}; // android only filter
>         await App.adapter.StartScanningForDevicesAsync(scanFilterOptions);
> ```

What error do you get?!? I think the line with the ServiceUuids may be problematic. Maybe you should not use so many filter criteria, otherwise you may not get any results. Start with a simple filter (addresses only?), and extend it later (after you have verified that it works).

yeohcl99, 2022-04-13:
ScanFilterOptions() not found.

janusw, 2022-04-13:
> ScanFilterOptions() not found.

Ah. Of course the current version (2.1.2) from nuget.org does not support this feature yet. I will prepare a new release soon. Until then you can use the nuget that is built from current master, see https://github.com/xabre/xamarin-bluetooth-le/actions/runs/2153569072 (scroll down to "Artifacts").

janusw, 2022-05-01:
The scan filtering was extended in #578 and #588.

yeohcl99, 2022-05-03:
[image]
I am still getting this error on 2.1.3

janusw, 2022-05-03:
The class you need to use is called `ScanFilterOptions` and it's in the namespace `Plugin.BLE.Abstractions`, see:

https://github.com/xabre/xamarin-bluetooth-le/blob/4b5a1065f0c17fae5a5b5890a59d69afe5e33b24/Source/Plugin.BLE.Abstractions/ScanFilterOptions.cs#L49