dotnet-bluetooth-le/dotnet-bluetooth-le #570: no devicediscover when screen is off at background service mode

yeohcl99, 2022-03-14:
adapter.DeviceDiscovered += (s, a) =>{} works well when the screen is on. But when screen is off on Android 9 the adapter.DeviceDiscovered += (s, a) =>{} no longer able to discover any near by BLE in background service mode. However when the screen comes on immediately able to scan the surrounding ble. Any idea how to resolve this issue?

## Steps to reproduce

1.  start adapter.devicediscovered running when screen on

2. turn screen off but start adapter.devicediscovered still running in background

3.detect no surrouding ble devices

## Expected behavior
adapter.devicedscovered should be able to scan surrounding ble devices in background servicemode be it screen on or off

## Actual behavior
adapter devicediscovered no surrounding device when screen is off

### Crashlog

NA

## Configuration

**Version of the Plugin:** 2.2 pre 2"

**Platform:** Android 9.1 pie 

**Device:** e.g. Samsung Galazy A32, Honor Play ...

janusw, 2022-03-14:
> adapter.DeviceDiscovered += (s, a) =>{} works well when the screen is on. But when screen is off on Android 9 the adapter.DeviceDiscovered += (s, a) =>{} no longer able to discover any near by BLE [...]

Duplicate of #565!

> Any idea how to resolve this issue?

With a ScanFilter, it should be possible to scan when the screen is off.

closed 2022-03-14 by janusw

yeohcl99, 2022-03-15:
i do not get what you meant by Scanfilter. Below is my code and when screen is off it will never able to detect the Rssi in my background service.

                    ble = CrossBluetoothLE.Current;//added ble
                    adapter = CrossBluetoothLE.Current.Adapter;//added ble
           
                    var state = ble.State;
                    var status = Permissions.RequestAsync<Permissions.LocationWhenInUse>();
                    adapter.ScanTimeout = 30000;
                    

            while (true)
            {
                try
                {
                    adapter.DeviceDiscovered += (s, a) =>
                    {
                        if (Int16.Parse(a.Device.Rssi.ToString()) > -140)
                        {
                            try
                            {
                                    // do something
                            }
                            catch (Exception ex) { }

                        }
                    };
                    if (!ble.Adapter.IsScanning)//default ble 
                    {
                        adapter.ScanMode = ScanMode.LowLatency;
                        await adapter.StartScanningForDevicesAsync();

                    }
                    Thread.Sleep(5000);

                }
                catch (Exception ex)
                {

                }

            }

janusw, 2022-03-15:
> i do not get what you meant by Scanfilter.

https://developer.android.com/reference/android/bluetooth/le/ScanFilter

Note that scan filtering is currently not supported in xamarin-bluetooth-le, see #515.

> Below is my code and when screen is off it will never able to detect the Rssi in my background service.

Yes, that is expected behavior (as described in #565).

janusw, 2022-05-01:
Note: Scan filtering on Android was extended with #578 and #588. This expands your options for achieving device discovery when the screen is off.