dotnet-bluetooth-le/dotnet-bluetooth-le #576: Do you have the stable version of Plugin.BLE for UWP?

leomurmur, 2022-03-24:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1.

2.

3.

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** e.g. HTC Sensation /i Phone 7 ...

janusw, 2022-04-08:
No, for now there is only the unstable pre-release 2.2.0-pre2 (https://www.nuget.org/packages/Plugin.BLE/2.2.0-pre2). There might be stable version for UWP in the future, but I personally cannot make any promises in this direction right now.

@leomurmur Have you tried 2.2.0-pre2? And if yes, does it work for you, or do you see any problems?

closed 2022-04-08 by janusw

leomurmur, 2022-10-11:
Hello,

I do have the problem using 2.2.0 –pre4. For UWP.  Function: GetSystemConnectedOrPairedDevices() returns device list count = 0, despite the fact there are paired devices by system.
Scanning also doesn’t work. The package 2.1.3 works pretty good for Android and iOS. Do I need to do something in additional to use 2.2.0- pre4, for UWP?
Thanks.

Regards,