dotnet-bluetooth-le/dotnet-bluetooth-le #577: Unable to Read Manufacturer Name

HarrisonOfTheNorth, 2022-03-24:
I'm trying to read the manufacturer's ```Name``` but come up with an empty string on the standard GATT ```2a29``` ```StringValue``` for the standard Device Information Service (```0000180a-0000-1000-8000-00805f9b34fb```), and byte array of zero elements for the ```Value``` of the same, however do get the Name showing as "Manufacture Name String".

## Steps to reproduce

1. I get the ```Device Information Service```, vis-a-vis:

```
            var serviceList = await device.GetServicesAsync();

            foreach (var service in serviceList)
            {
                if (service.Id == sigDeviceInformationServiceUuid) // i.e. 0000180a-0000-1000-8000-00805f9b34fb
                {
                    sigDeviceinformationService = service; // On breakpoint, this works, so far so good
                }
            }
```

2. We then get the Manufacturer GATT (```2A29```), vis-a-vis:

```
            if (sigDeviceinformationService != null)
            {
                var manufacturerGatt = await sigDeviceinformationService.GetCharacteristicAsync(sigManufacturerGattUuid); // i.e.  00002a29-0000-1000-8000-00805f9b34fb

                // Breakpoint on manufacturerGatt.Name shows "Manufacturer Name String"
                // Breakpoint on manufacturerGatt.StringValue shows ""
                // Breakpoint on manufacturerGatt.Value shows byte[0]
            }
```

3. The LightBlue app shows the actual Manufacturer Name String to be advertised, and I also verified this with another Nuget plugin (plugin.bluetoothLE) but this is the one I want in production

## Expected behavior
I expected ```manufacturerGatt.StringValue``` to show ```[Insert Manufacturer Name Here]```

## Actual behavior
A Breakpoint on manufacturerGatt.Name shows "Manufacturer Name String"

A Breakpoint on manufacturerGatt.StringValue shows ""

A Breakpoint on manufacturerGatt.Value shows byte[0]

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:** Android 12

**Device:** Samsung Galaxy S20+

Ignasrocas1990, 2022-03-30:
I started getting same problem.

janusw, 2022-04-07:
When you say "Android 12", is that the target framework version or the version of the device you're running it on, or both?

HarrisonOfTheNorth, 2022-04-07:
Hi @janusw thanks for replying! We resolved this yesterday, as things turned out, we had an all-eyes-on session and found that the API was badly written, as it turns out, between:

                var manufacturerGatt = await sigDeviceinformationService.GetCharacteristicAsync(sigManufacturerGattUuid); // i.e.  00002a29-0000-1000-8000-00805f9b34fb

and the breakpoints:

                // Breakpoint on manufacturerGatt.Name shows "Manufacturer Name String"
                // Breakpoint on manufacturerGatt.StringValue shows ""
                // Breakpoint on manufacturerGatt.Value shows byte[0]

you have to call:

               var iDontNeedThisReturnByteArray = await manufacturerGatt.ReadAsync();

even though you don't need the return value.

The really surprising thing is that in 14 days, no regular user picked this up, I'm not surprised that there are 251 forks of this, we're going to fork it as well so that we can support it internally without having to endure timeframes that are outside the scope of commercial normality.

I'm not knocking anyone here, just acknowleging that it's unsupported and from a commercial perspective, quite suspisciously in disarray.

It's nice that you got involved, though, thanks.

But solved now!

closed 2022-04-07 by HarrisonOfTheNorth

janusw, 2022-04-08:
> The really surprising thing is that in 14 days, no regular user picked this up, I'm not surprised that there are 251 forks of this, we're going to fork it as well so that we can support it [...]

Weeell, what can I say. First off, I don't like your tone. And I'm afraid there is a misunderstanding on your side (probably several). This is a community-driven open source project. There is no kind of 'commercial' support whatsoever. There are no guarantees that your bug report will be answered or even fixed within 48 hours. And AFAICS no one made that claim in the first place. So please update your skewed expectations. If you need commercial support, you'll have to pay someone to provide it.

That said, this certainly isn't the best-maintained project on GitHub, but it is neither "unsupported" nor "unmaintained". The way to improve the state of the project is not to go and create yet another fork that no one cares about except yourself, but to upstream the solutions to the problems you found, so that everyone can benefit from it, and we can keep developing one consistent version of the project instead of fraying it out.

Also please note that you will not be able to blackmail anyone by saying "Uuaahh. I'm going to fork!". You're just cutting your own flesh, if you depend on this project being vital. In the long run, you'll be better off contributing to the community than segregating.

And one more side note: A large number of forks is not necessarily a bad thing. On GitHub, creating a fork is the first step to making a contribution, therefore many forks can also be an indicator for a strong community involvement (as long as these forks are merged back into the original project, and don't become permanent "forks" in the literal sense).

Finally, sorry for giving you "the speech", but you seemed very needy in that respect ;)

HarrisonOfTheNorth, 2022-04-10:
@janusw 

You've misunderstood my tone to the extend that you assumed something that did not exist, and the rest of your assumptions were not valid, I understood the whole context.  

I also thanked you for your involvement, I don't know who you are or what your connection is, I'm a veteran with over 22 years in the mobile space, and my views are shared by other professionals in the industry with similar experience.

The value of me making my comment was more so that other observers could see the difficulties, so that they go in with their eyes open.

For example, rather than you saying "oh, you just need to call characteristic.ReadAsync() and ignore the return value, before calling characteristic.property, you instead asked me "When you say 'Android 12', is that the target framework version or the version of the device you're running it on, or both?" ... you didn't even spot the obvious resolution, which is not a criticism of you and I thank you for your attempt to help, but shows that even first responders aren't equipped with the API basics. 

- Let's face it, in the cold light of day, reading devices, then services, then characteristics, are the hello-world part of the API.
- The experience here is that there aren't any first responders available to answer simple 'hello characteristic' questions.

As you've decided to be harsh I'll add a little bit more context for observers. 

The only reason I made a comment in the first place was that I observed that there was consistent recent activity in the comments of the issues that were typically every 8 hours, so we expected someone to say something interesting next day. 

What actually happened was that the first response in 14 days even missed the obvious hello-world (i.e. "Hello Characteristic") issue. 

- We're not complaining, far from it, just giving a clear and concise agile retrospective opinion on the issue - which is what every single commercial and volunteer agile team does every sprint, world-wide.

Also, in that 14 days, we found it was quicker to write Android Bluetooth code from scratch without the use of the plugin. I bashed it out in 3 hours. Working. Wham, bam, thank-you maam.

Our assesement is effectively that this plugin is useful for rapid development but if one has the time, write one's own Android and iOS code directly is a good strategic opportunity. We are, however, pushing forwards with the plugin, doing so in an evironment where all critiics accept that there is not single C# bluetooth pluging that is not painful to utilise. That's a good assesement of the common view. But don't mistake this a s a criticism, but a valid agile retrospective comment.

We're thankful for all the developers who have contributed so far and on this occasion just offer our professional 'cold-light-of-day' retrospective - which is what agile is all about - for the benefit of others.

I've got a further Hello World example that I can post here that recursively drills down devices, services, and characteristics, looking for a known Manufacture Name of one's choosing.

But I'm not going to post it if I am further insulted.

It is a professional characteristic for professional engineers to post helpful retrospective comments to assist other engineers, and that is what I have done.

janusw, 2022-04-10:
> You've misunderstood my tone to the extend that you assumed something that did not exist, and the rest of your assumptions were not valid, I understood the whole context.

Alright then, dude, let's just file this under "misunderstanding" and move on. I really don't think it's worth to continue this discussion.

HarrisonOfTheNorth, 2022-04-10:
Yup, that's cool. I've got some demos coming that should help newbies.