dotnet-bluetooth-le/dotnet-bluetooth-le #579: Android 12 Device Discovery Issue and More

gtrigdon, 2022-04-01:
I have an Xamarin App using the Plugin that works perfectly in Android 11 but not Android 12.

As a side note, is there any way to use the Plugin API to get pairing status? In a parallel project I observed the following Android 12 behavior.

A12 appears to allow a phone to register the Services right away before pairing. This will problematic for the App because after connection it starts looking for the Services and then after registering them it moves forward with final secure connection to the target device. If the Secure message sequence occurs BEFORE the phone has paired with the device we have a problem. Ideally, if there was a way for the App to know whether or not pairing has occurred it could wait for pairing to complete before moving forward with the Secure message sequence to the target device.

This isn't a problem in A11 because the Services registration doesn't seem to work until after pairing. So if you are just waiting for registering Services all is fine because it doesn't happen until after pairing. A12 seems to be different and by not restricting Services before pairing, the App has no way of knowing when to proceed with further BLE messaging to the target device after the registration unless it has the pairing status.

## Steps to reproduce

1. Use Android 12

2.  await _bluetoothAdapter.StartScanningForDevicesAsync(null, null, false, source.Token);

3. 

## Expected behavior
I am expecting device discovery

## Actual behavior
For Android 12 I never seem to see any devices discovered, 

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** 2.12

**Platform:** Android 12

**Device:** Samsung S21

janusw, 2022-04-06:
About the discovery issue ...
You'll need to use the new Bluetooth permission scheme when targeting Android 12: https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

janusw, 2022-04-08:
> About the discovery issue ... You'll need to use the new Bluetooth permission scheme when targeting Android 12: https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

#582 shows how to do that (for the BLE.Client.Droid sample app). @gtrigdon, can you confirm that using the proper permissions fixes your issue with device discovery?

gtrigdon, 2022-04-08:
Hi, I figured out that the discovery issue was linked to the "Location" setting. If that setting is "off" I don't get discovery. I downloaded nRF connect App from Nordic and saw the same behavior.

Do you have any further feedback on the other issue revolving around pairing?

Thanks
Gerry

janusw, 2022-04-08:
> Hi, I figured out that the discovery issue was linked to the "Location" setting. If that setting is "off" I don't get discovery. I downloaded nRF connect App from Nordic and saw the same behavior.

You mean the location _permission_? Or what kind of 'setting' are you talking about?

Actually up to Android 11 (API levels <= 30), the location permission was required for using Bluetooth functionality. In Android 12 (API level 31) this was changed. A new permission scheme was introduced, and the location permission is not strictly required any more for using Bluetooth (for details see the link above).

Are you actually targeting Android 12 (API 31) with your app, or just running an app that targets Android 11 or lower on an Android 12 device?

> Do you have any further feedback on the other issue revolving around pairing?

Not really, sorry. I'm not quite sure I understand the problem. Also this should better be in a separate issue, since it's a completely different matter.

GO74, 2022-11-04:
> 

> Actually up to Android 11 (API levels <= 30), the location permission was required for using Bluetooth functionality. In Android 12 (API level 31) this was changed. A new permission scheme was [...]

Hi janusw, I stumbled over the same problem. I did implement exactly as shown in your sample. I could discover devices with my Android 11 device, but not in my Android 12 device. I then checked to see if the nRF Connect-App from Nordic was able to discover devices - they asked for 'ACCES_FINE_LOCATION'-Permission. So I gave it a try and added this request in my Android 12 code and it did the trick. Not sure what's going on here, but the problem is solved.

Thank you guys again for your awesome job. I'm using this lib now for over 5 years - it's great!

[image]