dotnet-bluetooth-le/dotnet-bluetooth-le #580: Can't connect to BLE-device when compiling with Android 12

GO74, 2022-04-05:
## Steps to reproduce

1. Build for Android target 12

2. Deploy to Pixel2 Android 11/Samsung S5 Android 9

3. Connect to BLE-Device (await bleAdapter.ConnectToDeviceAsync(device, cp, ct);)

## Expected behavior
Should be successfully connect to BLE-Device in both cases, but it only works on the Android 9 Samsung device.
It previously worked on Pixel2 Android11 when target version and target framework was 10.

## Actual behavior
Connection to BLE-Device on Pixel2 Android 11 never returns, cancellation token takes over.

### Crashlog
[BluetoothGatt] connect() - device: D0:CF:5E:32:08:A1, auto: false
[BluetoothGatt] registerApp()
[BluetoothGatt] registerApp() - UUID=7106bc0a-1059-4c24-9656-b77f09cab387
[BluetoothGatt] onClientRegistered() - status=0 clientIf=8
[BluetoothGatt] cancelOpen() - device: D0:CF:5E:32:08:A1
[BluetoothGatt] close()
[BluetoothGatt] unregisterApp() - mClientIf=8

## Configuration
Plugin.BLE(2.1.2)
Targetframework 12.0 (S)
Targetversion 12.0 (S)
Android minimum 4.2

Access_coarse_location
Access_fine_location
Bluetooth
Bluetooth_Admin

**Platform:** 
Android 11

**Device:**
Pixel2 Android 11/Samsung S5 Android 9

janusw, 2022-04-06:
> Access_coarse_location Access_fine_location Bluetooth Bluetooth_Admin

This may be a permission problem. The Bluetooth permissions you quote are the old ones. When targeting Android 12, you need to use the new Bluetooth permission scheme in principle: https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

However, when running on older Android devices, you need to supply the old permissions for backward compatibility.

Did I get you right: If you target Android 12, it works when running on Android 9, but not on Android 11?

GO74, 2022-04-06:
"Did I get you right: If you target Android 12, it works when running on Android 9, but not on Android 11?"
Yes, I did change target and deployed to Anroid 9 and Android 11. Android 9+11 both discover devices, but 11 isn't able to connect. I'll check out the permissions.

GO74, 2022-04-07:
I had to update the issue because after compiling for Android 11 I experienced the same problem. I have to assume that this is not a permission issue as this would only be a problem when targeting Android 12.

janusw, 2022-04-07:
> I had to update the issue because after compiling for Android 11 I experienced the same problem. I have to assume that this is not a permission issue as this would only be a problem when targeting Android 12.

Ah, interesting, so I guess it must be something different.

We have another bug report regarding Android 11 (#558), but there it was claimed that already scanning does not work with Android 11 (which I could not reproduce). Scanning works in your case, and only connecting fails, right?

Could you try the BLE.Client.Droid app, in order to check if your problem occurs there as well? It is targeting Android 11 on the current master branch ...

GO74, 2022-04-07:
Hey Janus, I just updated my reply from a minute ago. Any suggestions on how to proceed are welcome. I'll now try to install BLE.Client.Droid app on the Pixel2 device.

GO74, 2022-04-07:
Okay, we can close this issue. ALWAYS TRY TURNING IT OFF AND ON AGAIN.
I just rebooted the Pixel2 device and it's now working fine on all devices with Android 11 as target.

closed 2022-04-07 by GO74

janusw, 2022-04-07:
> Okay, we can close this issue. ALWAYS TRY TURNING IT OFF AND ON AGAIN. I just rebooted the Pixel2 device and it's now working fine on all devices with Android 11 as target.

Ah, yeah. That's a classic 🤣

https://memegenerator.net/img/instances/85585319/have-you-tried-turning-it-on-and-off-again.jpg

Thanks for clearing this up, though 👍

GO74, 2022-04-07:
It's very good to know that you guys are around to help if you can. Vielen Dank!