dotnet-bluetooth-le/dotnet-bluetooth-le #581: Why location permission is mandatory?

rajnikantpatel23, 2022-04-06:
Hi, Can someone please explain why Location permission is mandatory for this nuget to work. Does it improve accuracy for pairing? Does it make pairing faster? Or some other benefit?

closed 2022-04-06 by rajnikantpatel23

janusw, 2022-04-06:
> Hi, Can someone please explain why Location permission is mandatory for this nuget to work. Does it improve accuracy for pairing? Does it make pairing faster? Or some other benefit?

Well, this requirement is not specific to this project. It is dictated by the Android OS. Also it has nothing to do with improving or speeding up the pairing process. AFAIK it was introduced in Android, because one can in principle use Bluetooth data to deduce location. And in fact the requirement has been lifted in Android 12, with which it is possible now to use Bluetooth without the location permission.

janusw, 2022-04-06:
See also #568.