dotnet-bluetooth-le/dotnet-bluetooth-le #583: GitHub Actions: run unit tests

janusw, 2022-04-09:
The GitHub-Actions CI build currently does the following: It builds the lib and the client app and generates nuget packages. One thing that is still missing is running the unit tests (XUnit-based).

closed 2022-04-11 by janusw

SSstupid, 2022-07-26:
Hi, janusw
How to unit test, Can i do that?
I trying to unit test bluetooth library, It made from Plugin.Ble
Run XUnit test code Then i see this error message
=> Error : This functionality is not implemented in the portable version of this assembly.  You should reference the NuGet 
package from your main application project in order to reference the platform-specific implementation.

I think value is null.....

janusw, 2022-07-26:
> Hi, @janusw How to unit test, Can i do that?

Sure, you could either run them directly via Visual Studio, or through the cake hooks that were added here: e2299407a8197f23c3611e52e2f82f2bae7ab395. To build and run the tests via cake, just do this on the command line:
```
cake --target=BuildTests
cake --target=RunTests
```
HTH.

SSstupid, 2022-07-27:
> Sure, you could either run them directly via

Hi @janusw 
I don't understand. Can you tell me in detail?
How to run via Visual Studio?,
What is cake hooks? Is it IDE or something else?

janusw, 2022-07-27:
> Hi @janusw I don't understand. Can you tell me in detail? How to run via Visual Studio?

With VS for Mac, you just open BLE.Mac.slnf and then choose "Debug -> Run Unit Tests" from the main menu. There is also a "Tests" window in the sidebar that you can use to run and inspect single tests. In VS for Windows, it possibly works a bit different (I don't use that a lot).

> What is cake hooks? Is it IDE or something else?

cake is a build system for .NET that you typically use from the terminal (see commands above). For more info see https://cakebuild.net. There are also extensions to integrate it with various IDEs (VS, VS Code, ...).

PR #584 (2022-04-11): GitHub Actions: run tests
This extends the GHA config to build & run the unit tests. As a preparation, the test project is updated to .NET 5 and the cake script is extended to support the unit test. GHA runs the corresponding cake targets in the end. Fixes #583.