dotnet-bluetooth-le/dotnet-bluetooth-le #587: Error on iOS after ConnectToDeviceAsync()

jerkohren, 2022-04-14:
Hallo, my implementation of the library on Android works as expected. However, after I transfer to iOS, I have expirienced following problem:

## Steps to reproduce

1. ConnectToDeviceAsync()

## Expected behavior
Return the list of WiFi available (nymea network manager as bluetooth server) or any kind of data

## Actual behavior
App doesn't receive any data from connected bluetooth device.

### Crashlog

Apr 14 11:26:44 reefmasterpi nymea-networkmanager[1035]:  I | NetworkManagerBluetoothServer: Controller state advertising...
Apr 14 11:26:44 reefmasterpi nymea-networkmanager[1035]:  I | NetworkManagerBluetoothServer: Set running true
Apr 14 11:26:44 reefmasterpi nymea-networkmanager[1035]:  I | NetworkManagerBluetoothServer: Bluetooth host in discoverable mode.
Apr 14 11:26:44 reefmasterpi nymea-networkmanager[1035]:  I | Application: Bluetooth server started
Apr 14 11:28:33 reefmasterpi nymea-networkmanager[1035]:  I | NetworkManagerBluetoothServer: Device connected "59:FB:58:90:45:CB"
Apr 14 11:28:34 reefmasterpi nymea-networkmanager[1035]:  I | NetworkManagerBluetoothServer: Controller state connected. "59-FB-58-90-45-CB" "59:FB:58:90:45:CB"
Apr 14 11:28:34 reefmasterpi nymea-networkmanager[1035]:  I | NetworkManagerBluetoothServer: Set connected true
Apr 14 11:28:34 reefmasterpi nymea-networkmanager[1035]:  I | NetworkManagerBluetoothServer: Client connected "59-FB-58-90-45-CB" "59:FB:58:90:45:CB"
Apr 14 11:28:34 reefmasterpi nymea-networkmanager[1035]:  I | Application: Bluetooth client connected
Apr 14 11:28:34 reefmasterpi nymea-networkmanager[1035]:  W | qt.bluetooth.bluez: sending error response; request: 16 handle: 51 code: 10

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:**  iOS 14.4.2

**Device:** iPhone 8

Do you have some idea what could be wrong?

janusw, 2022-04-14:
Hard to tell what happens without seeing any of you code :/

The error message you post is an error on the server side, apparently. Do you have an idea what it means? What is code 10?

Do you get any errors on the client side? What happens there? What is the last thing the client does?

In principle it might just as well be a bug in the server ...?

jerkohren, 2022-04-15:
Well the code error is comming from here: [(https://code.woboq.org/qt5/qtconnectivity/src/bluetooth/qlowenergycontroller_bluez.cpp.html#8handle)]
code 10 is ATT_ERROR_ATTRIBUTE_NOT_FOUND but i don't know what this mean, should I for iOS connect with some paramaters?