dotnet-bluetooth-le/dotnet-bluetooth-le #590: Different device ids for Android and iOS

birkneral, 2022-04-20:
I'm currently working on an app with multiple Bluetooth functions.

One of these functions should be the display of the ID of found devices to identify them. Realizing this, I noticed that the Id property of the device class contains different IDs when using Android or iOS. In the device classes for Android and iOS, I found that the ids are obtained from different sources and are therefore probably also parsed differently.

Shouldn't the Device.Id property contain the same value regardless of the operating system? Or are there any operating system-related reasons against it? It would be great if there is a way to show the same Id for the same devices on Android and iOS.

Thank you for your support and the plugin you created.

## Steps to reproduce

1. Search for devices

2. Display the property Device.Id for any found BLE device on a Android and a iOS mobile device

## Expected behaviour
The same id is displayed.

## Actual behaviour
Different ids are displayed.

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:**  Android 11 /  iOS 15.3.1

**Device:** Samsung Galaxy S21 5G / iPhone SE

janusw, 2022-04-25:
> I'm currently working on an app with multiple Bluetooth functions.
>
> One of these functions should be the display of the ID of found devices to identify them. Realizing this, I noticed that the Id [...]

To my knowledge this is simply due to OS limitations: iOS and Android just give you different kinds of IDs. While on Android you basically get the MAC address as ID, iOS does not provide the MAC but some kind of UUID (see e.g. https://developer.apple.com/forums/thread/8442).

birkneral, 2022-04-29:
Many thanks for the answer.

closed 2022-04-29 by birkneral