dotnet-bluetooth-le/dotnet-bluetooth-le #591: Does this work on UWP? If not why does the documentation say it does? How can I contribute to making UWP work?

nchrisr, 2022-04-22:
Does this work on UWP? If not why does the readme say it does? 

How can I contribute to making UWP work? 

Is there a specific branch on which this is being worked on so that I can help out?

## Steps to reproduce
Try using it on UWP

## Expected behavior
Should work

## Actual behavior
Doesn't work

## Configuration

**Version of the Plugin:** 2.1.2

**Platform:** UWP

janusw, 2022-04-23:
> Does this work on UWP? If not why does the readme say it does?

Well, the stable version 2.1.2 does not support UWP, but there is a 2.2.0 pre-release with some amount of UWP support (which is not fully 'final' yet). I haven't tried it myself, and therefore I'm not sure what state it is in.

> Is there a specific branch on which this is being worked on so that I can help out?

https://github.com/xabre/xamarin-bluetooth-le/commits/uwp-support

> How can I contribute to making UWP work?

Check the current state of things, identify things that do no work. Document them and/or open issues (if not existing yet). Possibly try to fix them (open a PR if you succeed).

baskren, 2022-04-25:
@nchrisr  : Let me know if you want to collaborate on this.

nchrisr, 2022-04-25:
@baskren I would be interested in collaborating on this. I will need some time to have a look at the architecture and see what I can decypher, but based on your description, I have found some links that may be useful 

- https://github.com/jasongin/noble-uwp/issues/20#issuecomment-371409715
- https://stackoverflow.com/questions/39599252/windows-ble-uwp-disconnect/43609852#43609852

Let me know how you want us to proceed.

baskren, 2022-04-25:
@nchrisr :  Yes, I think we're on the same page (with respect to what's needed to disconnect).  The code isn't quite architected to make this quick, one line change.  I'll take a look and post for you a proposed change.  I don't think it's going to be pretty so it would be nice to get some feedback from @janusw as to how he would like this to be done (so I don't have to face the crippling assault on my self esteem when he - for good reason - rejects a half baked pull request).

baskren, 2022-04-25:
FWIW, I was able to get the `uwp_creators_support` branch working by making the following changes:

**Source/Plugin.BLE.UWP/Device.cs**
Add the following:
```csharp
        public void NativeDisconnect()
        {
            ClearServices();

            var bluetoothAddress = Convert.ToUInt64(Id.ToString("N").Substring(20), 16);
            NativeDevice.BluetoothLEDevice.Dispose();
            Task.Run(async () =>
            {
                var nativeDevice = await BluetoothLEDevice.FromBluetoothAddressAsync(bluetoothAddress);
                NativeDevice = new ObservableBluetoothLEDevice(nativeDevice.DeviceInformation);
            });

        }
```

**Source/Plugin.BLE.UPW/Adapter.cs**
change the contents of **DisconnectDeviceNative** method to:
```csharp
        protected override void DisconnectDeviceNative(IDevice device)
        {
            // Windows doesn't support disconnecting, so currently just dispose of the device
            Trace.Message($"Disconnected from device with ID:  {device.Id}");
            /*
            if (device.NativeDevice is ObservableBluetoothLEDevice)
            {
                ((Device)device).ClearServices();
                device.Dispose();                
            }
            */

            if (device is UWP.Device uwpDevice)
                uwpDevice.NativeDisconnect();
        }
```

**Source/Plugin.BLE.Abstractions/AdapterBase.cs**
change the content of **DisconnectDeviceAsync** to:
```csharp
        public Task DisconnectDeviceAsync(IDevice device)
        {
            if (!ConnectedDevices.Contains(device))
            {
                Trace.Message("Disconnect async: device {0} not in the list of connected devices.", device.Name);
                return Task.FromResult(false);
            }

            /*
            return TaskBuilder.FromEvent<bool, EventHandler<DeviceEventArgs>, EventHandler<DeviceErrorEventArgs>>(
               execute: () => DisconnectDeviceNative(device),

               getCompleteHandler: (complete, reject) => ((sender, args) =>
               {
                   if (args.Device.Id == device.Id)
                   {
                       Trace.Message("DisconnectAsync Disconnected: {0} {1}", args.Device.Id, args.Device.Name);
                       complete(true);
                   }
               }),
               subscribeComplete: handler => DeviceDisconnected += handler,
               unsubscribeComplete: handler => DeviceDisconnected -= handler,

               getRejectHandler: reject => ((sender, args) =>
               {
                   if (args.Device.Id == device.Id)
                   {
                       Trace.Message("DisconnectAsync", "Disconnect Error: {0} {1}", args.Device?.Id, args.Device?.Name);
                       reject(new Exception("Disconnect operation exception"));
                   }
               }),
               subscribeReject: handler => DeviceConnectionError += handler,
               unsubscribeReject: handler => DeviceConnectionError -= handler);
            */

            return Task.Run(() => DisconnectDeviceNative(device));
        }
```

Note that I'm not yet sure why I had to make that last change - for some reason, the `getCompleteHandler` is not fired when the ``execute:`` action is completed.  Haven't looked into that yet.

Also, I did all of the above _before_ I merged the latest `master` commit with the `uwp-support` branch.  Looks like there are some interface changes that the UWP implementation needs before a more thorough investigation (of the above issue) can occur.

baskren, 2022-05-02:
Wow. @janusw , if you could do #1, that would be most awesome!  Yes, count me in for #2 and #4.

janusw, 2022-05-02:
> Wow.

What was that for? The twelve open UWP issues?

> @janusw , if you could do #1, that would be most awesome!

I'll try to look into it within a week ...

> Yes, count me in for #2 and #4.

Awesome!

janusw, 2022-05-14:
> > I did not do any actual runtime testing yet. @baskren I hope you can take over for step #2 here ...
> 
> I have tested it with my code and the only problem I found so far was a null-reference crash in StartScanningForDevicesAsync (pull request #594 submitted).

Thanks for the testing! I have just merged that one into the `uwp` branch.

fozzzgate, 2022-05-14:
Just submitted a new PR - Was getting some more null-reference crashes during GATT Service queries. I can confirm that the proposed changes prevent the crashes, however, I am not sure that this is the best way to handle null-references that occur in async methods. Please review before merging.

As for the reason that Microsoft.Toolkit.Uwp.Connectivity versions >= 7.0.0 cause a crash, apparently it is because ObservableBluetoothLEDevice now uses the property HasThreadAccess, which was only introduced in Windows SDK version 10.0.18362.0 - However the NuGet package declares that it only requires Windows SDK 10.0.17763.0

fozzzgate, 2022-05-15:
Closed PR #595 (see comments for why) and submitted revised PR #596 to match the way that the Android code handles nulls in GetServicesNativeAsync()

janusw, 2022-05-24:
Thanks for your testing and improvements so far, @fozzzgate.

I have also pushed a few more changes to the `uwp` branch. The GitHub Actions build job runs through now (and produces a nuget that can be used for testing).

However, there is still one problem with the sample client (BLE.Client.UWP). It throws this exception on startup:
```
System.ArgumentException: '[Acr.UserDialogs] This is the bait library, not the platform library.  You must install the nuget package in your main executable/application project'
```
I am a bit puzzled about how to fix this. AFAICS the Acr.UserDialogs package is included both in BLE.Client and BLE.Client.UWP. Does anyone have an idea what to do about this?

nchrisr, 2022-05-25:
> > Ended up taking a quick look at the BLE.Client.UWP 'Scan' crash and pushed a fix as part of PR #599
> 
> Cool. This is merged by now, making the UWP sample client work finally. Also the GHA build works flawlessly on the UWP branch.
> 
> This represents sort of a small milestone to me, which justifies tagging a first pre-release of 2.2.0.
> 
> @nchrisr, @baskren Any feedback from your side at this point? Do you see any urgent showstoppers? Would you agree that it's a good time to tag a pre-release and upload it to nuget.org? Or is there anything that should be fixed first?

@janusw Apologies I have been a unavailable for a bit. I will test this out tomorrow and share any information on any issues I find.

janusw, 2022-07-11:
I just released 2.2.0-pre5, and I'm closing this issue. For any remaining problems with UWP, please open a new issue.

closed 2022-07-11 by janusw