dotnet-bluetooth-le/dotnet-bluetooth-le #592: MAUI Blazor

mykolapry, 2022-04-29:
Hello,
I try use Plugin.BLE with Maui Blazor.
With Androion all work fine but I have problem with IOs.
Curren always return: "System.NotImplementedException: This functionality is not implemented in the portable version of this assembly.  You should referenc…"

## Steps to reproduce

1. using Plugin.BLE;
    using Plugin.BLE.Abstractions.Contracts;
    using Plugin.BLE.Abstractions.EventArgs;
    using Plugin.BLE.Abstractions.Exceptions;

2. protected override async Task OnParametersSetAsync()
        {
            var ble = CrossBluetoothLE.Current;
        }

3. Info.plist keys were added

## Expected behavior
Get IBluetoothLE obj

## Actual behavior
Curren always return: "System.NotImplementedException: This functionality is not implemented in the portable version of this assembly.  You should referenc…"

## Configuration

**Version of the Plugin:** e.g. 2.1.2 / 2.2.0-pre

**Platform:** e.g. iOS 14.2 and more

**Device:** iPhone 12mini

janusw, 2022-04-29:
> Hello, I try use Plugin.BLE with Maui Blazor. With Androion all work fine but I have problem with IOs. Curren always return: "System.NotImplementedException: This functionality is not implemented in the portable version of this assembly. You should referenc…"

This exception is `CrossBluetoothLE.NotImplementedInReferenceAssembly`, see https://github.com/xabre/xamarin-bluetooth-le/blob/d873642e0858fe4a21e3e4d77f8d50a5735df4a7/Source/Plugin.BLE/CrossBluetoothLE.cs#L40

I haven't tried the library with MAUI yet. Do you have a reproducer project?

mykolapry, 2022-04-29:
No, I have only base maui blazor project and I try this in the method on any page
protected override async Task OnParametersSetAsync()
{
var ble = CrossBluetoothLE.Current;
}

smstuebe, 2022-04-29:
For maui we need to rebuilt the library against .net 6 as I'm currently doing on my other plugin: 
WIP: https://github.com/smstuebe/xamarin-fingerprint/commits/maui-support

But regarding blazor I don't think this will help, because this plugin simply doesn't support blazor (at least not officially)

mykolapry, 2022-04-29:
If you have fix for maui we can try it for maui blazor

danijerez, 2022-04-30:
@mykolapry could you give an example of how you use it in Maui Blazor Android? I'm trying but can't find any device

mykolapry, 2022-05-01:
> @mykolapry could you give an example of how you use it in Maui Blazor Android? I'm trying but can't find any device

Hello,  yes
using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;
using Plugin.BLE.Abstractions.EventArgs;

namespace TestMobileBlazor.Pages
{
    public partial class SomeClass
    {
        public IAdapter _adapter;      
        private List<IDevice> _bluetooth = new List<IDevice>();

        protected override async Task OnParametersSetAsync()
        {

            var adapter = CrossBluetoothLE.Current.Adapter;
            _adapter = CrossBluetoothLE.Current.Adapter;
            _adapter.DeviceDiscovered += DeviceDiscovered;

            await Scan();

            await InvokeAsync(StateHasChanged);
        }

       
        private async Task Scan()
        {
           
            await ScanBle();
        }

        private async Task ScanBle()
        {
                await _adapter.StartScanningForDevicesAsync();           
        }

        public async void DeviceDiscovered(object s, DeviceEventArgs device)
        {
            if (device.Device.Name == null)
                return;

            _bluetooth.Add(device.Device);
           // Console.WriteLine(device.Device.Name);
        }
    }
}

and in the AndroidManifest.xml
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionCode="1" android:versionName="1.0" package="com.mint.dispatchMobile.Droid">
  <uses-sdk android:minSdkVersion="21" android:targetSdkVersion="30" />
  <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
  <uses-permission android:name="android.permission.BLUETOOTH" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  <uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.ACCEPT_HANDOVER" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
  <uses-permission android:name="android.permission.WRITE_SETTINGS" />
  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true" /> 
</manifest>

mykolapry, 2022-05-01:
> @mykolapry could you give an example of how you use it in Maui Blazor Android? I'm trying but can't find any device

and in the AndroidManifest.xml

  <uses-sdk android:minSdkVersion="21" android:targetSdkVersion="30" />
  <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
  <uses-permission android:name="android.permission.BLUETOOTH" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  <uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.ACCEPT_HANDOVER" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
  <uses-permission android:name="android.permission.WRITE_SETTINGS" />
  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />

danijerez, 2022-05-01:
@mykolapry Thanks! the code is more or less what I used and I can't find any device either, I'm doing tests with an Android 11 ​​and 12 mobile. the bluetooth is an ab1602 from a robot penguin elegoo arduino.

mykolapry, 2022-05-01:
> @mykolapry Thanks! the code is more or less what I used and I can't find any device either, I'm doing tests with an Android 11 ​​and 12 mobile. the bluetooth is an ab1602 from a robot penguin elegoo arduino.

Are you publish it on real device or in simulator? 
Because it's working only on real device.

danijerez, 2022-05-01:
real device. I connect them by usb, I have them in debug mode and I execute the code sending it from vs 2022 preview

danijerez, 2022-05-01:
> Are you publish it on real device or in simulator? Because it's working only on real device.

 real device. I connect them by usb, I have them in debug mode and I execute the code sending it from vs 2022 preview

mykolapry, 2022-05-01:
> real device. I connect them by usb, I have them in debug mode and I execute the code sending it from vs 2022 preview
Yes, the same. I will double check this tomorrow and send you all solution, because I don't have device today at home.

danijerez, 2022-05-01:
> > real device. I connect them by usb, I have them in debug mode and I execute the code sending it from vs 2022 preview
> > Yes, the same. I will double check this tomorrow and send you all solution, because I don't have device today at home.

OK thank you very much. I don't know what I'm doing wrong, I can't find much documentation about the library and bluetooth in general.

mykolapry, 2022-05-02:
> OK thank you very much. I don't know what I'm doing wrong, I can't find much documentation about the library and bluetooth in general.
I sent to email solution example

danijerez, 2022-05-02:
@mykolapry thank you! I'm reviewing it, although at first glance I don't see anything different, do you manage to detect bluetooth devices?

danijerez, 2022-05-02:
https://stackoverflow.com/questions/50509591/plugin-ble-for-xamarin-forms-not-detecting-devices 

It is possible that something similar happens, I can't find how to check the permissions in maui of:

Manifest.Permission.Bluetooth,
Manifest.Permission.BluetoothAdmin,
Manifest.Permission.AccessCoarseLocation,
Manifest.Permission.AccessFineLocation

busec0, 2022-06-14:
Any updates on this? Are you planning to add support for MAUI & .NET 6 anytime soon?

jamesmontemagno, 2022-08-10:
@busec0  why not send down a PR? We should work together on this ;)

busec0, 2022-08-31:
Unfortunately, mostly due to time constraints, but I see you were on top of this already https://github.com/xabre/xamarin-bluetooth-le/pull/614!

jamesmontemagno, 2022-09-01:
I'm getting closer into the official package here.

TimoKinnunen, 2022-09-16:
Bluetooth LE is a headache, take a look here https://github.com/TimoKinnunen/MauiXAMLBluetoothLE. This is a good starting point to bring Bluetooth LE (Plugin.BLE) closer to .Net MAUI, I think.

janusw, 2022-11-15:
It was reported that #638 fixes the problems with MAUI on iOS (see also the discussion in #614). @mykolapry, can you confirm that your problem is gone on current master? (using the nupkg from here: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/actions/runs/3470893880#artifacts)

janusw, 2022-11-15:
> **LE**: For anyone interested, see https://www.nuget.org/packages/Plugin.BLE.Net/3.0.0-alpha.1 I forked the repository and updated the target to net6, but only for iOS and Android for now.

@busec0 General advice: If you have any kind of improvement for this library, please open a PR, instead of publishing your own nugets. You will get more feedback and testing this way, and if we can agree on a common version, this will be more useful to everyone (as opposed to forking off custom versions that are never merged back).

busec0, 2022-11-15:
@janusw Makes sense, it was just a quick update to .Net 6 for anyone needing that sooner to play around with MAUI / Blazor. No actual improvements on my side, just the retargeting. I see you've brought .NET 6 support now so I'll remove the package. Thanks!

janusw, 2022-11-15:
> @janusw Makes sense, it was just a quick update to .Net 6 for anyone needing that sooner to play around with MAUI / Blazor. No actual improvements on my side, just the retargeting. I see you've brought .NET 6 support now so I'll remove the package. Thanks!

Thanks for understanding :)

If you have any feedback on the current state of things with .NET6/MAUI, please let us know. Does it work for you now?

Btw, every PR you submit here also builds a nupkg automatically (which can be downloaded and used e.g. via a local package source), so that people can play around and test it. Really no need to upload such test versions directly to nuget.org, it rather tends to confuse people ;)

janusw, 2022-11-18:
So, I have released version [3.0.0-beta.1](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.0.0-beta.1) yesterday and pushed it to nuget.org.

@mykolapry @danijerez @busec0 Can anyone confirm that this works well with Blazor? Also on iOS? I'd love to hear some feedback here!

busec0, 2022-11-18:
@janusw Just tested on a MAUI .NET 7 iOS app, everything seems to be working fine!

janusw, 2022-11-18:
Yes, in general things seem to work well with MAUI on iOS. I've tried that myself already.

My question was more specifically whether things also work well in a Blazor context. I assume that is the case, but ideally I would like to have this assumption confirmed by someone who has actually tried it :)

danijerez, 2022-11-20:
I did a small test in maui blazor, with windows it detects the devices, although I get exceptions regarding CommunityToolkit.WinUI.dll, maybe it is something that I have configured wrong

in Android, adding the permissions, 

```
	<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
	<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
```

I can't get it to work, I'm testing it in Android 12 in pixel 2 xl and Android 11 emulated, I leave the example code

If I discover something I will comment
```
@page "/"
@using Plugin.BLE;
@using Plugin.BLE.Abstractions.Contracts;
@using Plugin.BLE.Abstractions.EventArgs;

@if (_bluetooth != null && _bluetooth.Count > 0)
{
    foreach (var b in _bluetooth)
    {
        <div>@b.Name</div>
    }
}

@code{

    public IAdapter _adapter;
    private List<IDevice> _bluetooth = new List<IDevice>();

    protected override async Task OnParametersSetAsync()
    {

        var adapter = CrossBluetoothLE.Current.Adapter;
        _adapter = CrossBluetoothLE.Current.Adapter;
        _adapter.DeviceDiscovered += DeviceDiscovered;

        await Scan();

        await InvokeAsync(StateHasChanged);
    }

    private async Task Scan()
    {
        await _adapter.StartScanningForDevicesAsync();
    }

    public void DeviceDiscovered(object s, DeviceEventArgs device)
    {
        if (device.Device.Name == null)
            return;
        _bluetooth.Add(device.Device);
    }

}
```

TimoKinnunen, 2022-11-20:
Hello Janus Weil and James Montemagno

This is about .NET Maui and Bluetooth LE on Windows:

Janus Weil and James Montemagno and others have done amazing work so far wih Plugin.BLE.

I have tried Plugin.BLE in Android emulator and in smartphone and it seems to be ok.
I have not possibility to test Plugin.BLE on iOS because I haven't Apple developer account yet. Janus Weil reports it seems ok.

And I found cumbersome Plugin.BLE on Windows. Could you take a look at it? Bluetooth LE devices are not discovered.

Download from https://github.com/TimoKinnunen/MauiXAMLBluetoothLE and play around with it!

<ItemGroup>
 <PackageReference Include="CommunityToolkit.Maui" Version="3.0.0" />
 <PackageReference Include="CommunityToolkit.Mvvm" Version="8.1.0-preview1" />
 <PackageReference Include="Plugin.BLE" Version="3.0.0-beta.1" />
</ItemGroup>

With regards Timo Kinnunen

janusw, 2022-11-24:
> I did a small test in maui blazor, with windows it detects the devices, although I get exceptions regarding CommunityToolkit.WinUI.dll, maybe it is something that I have configured wrong

@danijerez Some problems with MAUI on Windows were fixed in version [3.0.0-beta.2](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.0.0-beta.2), which I have just released. If you still get any exceptions with this, please report back here (preferably with a stack trace).

TimoKinnunen, 2022-11-24:
Using PackageReference Include="Plugin.BLE" Version="3.0.0-beta.2":
On Windows for example this does not work IReadOnlyList systemDevices = Adapter.GetSystemConnectedOrPairedDevices(HeartRateUuids.HeartRateServiceUuids);
Result is systemDevices = 0;
The good thing is no exceptions.

danijerez, 2022-11-24:
With version 3.0.0-beta.2 in windows it seems that it already detects the devices correctly.

I keep testing on Android 12.1, although I add the permission

```
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
```
the following exception is thrown
```
[monodroid-assembly] open_from_bundles: failed to load assembly System.Runtime.Intrinsics.dll
Loaded assembly: /data/data/com.companyname.testmobileble/files/.__override__/System.Runtime.Intrinsics.dll [External]
[monodroid-assembly] open_from_bundles: failed to load assembly Plugin.BLE.dll
Loaded assembly: /data/data/com.companyname.testmobileble/files/.__override__/Plugin.BLE.dll [External]
[BluetoothAdapter] isLeEnabled(): BLE_ON
[chromium] [INFO:CONSOLE(1)] "Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@bfcdc910: GattService registerScanner
[chromium]    at Java.Interop.JniEnvironment.InstanceMethods.CallVoidMethod(JniObjectReference instance, JniMethodInfo method, JniArgumentValue* args) in /Users/runner/work/1/s/xamarin-android/external/Java.Interop/src/Java.Interop/obj/Release/net7.0/JniEnvironment.g.cs:line 11884
[chromium]    at Java.Interop.JniPeerMembers.JniInstanceMethods.InvokeAbstractVoidMethod(String encodedMember, IJavaPeerable self, JniArgumentValue* parameters) in /Users/runner/work/1/s/xamarin-android/external/Java.Interop/src/Java.Interop/Java.Interop/JniPeerMembers.JniInstanceMethods_Invoke.cs:line 47
[chromium]    at Android.Bluetooth.LE.BluetoothLeScanner.StartScan(IList`1 filters, ScanSettings settings, ScanCallback callback) in /Users/runner/work/1/s/xamarin-android/src/Mono.Android/obj/Release/net7.0/android-33/mcw/Android.Bluetooth.LE.BluetoothLeScanner.cs:line 116
[chromium]    at Plugin.BLE.Android.Adapter.StartScanningNew(ScanFilterOptions scanFilterOptions) in D:\a\dotnet-bluetooth-le\dotnet-bluetooth-le\Source\Plugin.BLE\Android\Adapter.cs:line 200
[chromium]    at Plugin.BLE.Android.Adapter.StartScanningForDevicesNativeAsync(ScanFilterOptions scanFilterOptions, Boolean allowDuplicatesKey, CancellationToken scanCancellationToken) in D:\a\dotnet-bluetooth-le\dotnet-bluetooth-le\Source\Plugin.BLE\Android\Adapter.cs:line 60
[chromium]    at Plugin.BLE.Abstractions.AdapterBase.StartScanningForDevicesAsync(ScanFilterOptions scanFilterOptions, Func`2 deviceFilter, Boolean allowDuplicatesKey, CancellationToken cancellationToken) in D:\a\dotnet-bluetooth-le\dotnet-bluetooth-le\Source\Plugin.BLE\Shared\AdapterBase.cs:line 76
[chromium]    at TestMobileBle.Pages.Index.Scan() in C:\workspace\maui\TestMobileBle\Pages\Index.razor:line 34
[chromium]    at TestMobileBle.Pages.Index.OnParametersSetAsync() in C:\workspace\maui\TestMobileBle\Pages\Index.razor:line 27
[chromium]    at Microsoft.AspNetCore.Components.ComponentBase.CallStateHasChangedOnAsyncCompletion(Task task)
[chromium]    at Microsoft.AspNetCore.Components.ComponentBase.RunInitAndSetParametersAsync()
[chromium]   --- End of managed Java.Lang.SecurityException stack trace ---
[chromium] java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@bfcdc910: GattService registerScanner
[chromium] 	at android.os.Parcel.createExceptionOrNull(Parcel.java:2426)
[chromium] 	at android.os.Parcel.createException(Parcel.java:2410)
[chromium] 	at android.os.Parcel.readException(Parcel.java:2393)
[chromium] 	at android.os.Parcel.readException(Parcel.java:2335)
[chromium] 	at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerScanner(IBluetoothGatt.java:1779)
[chromium] 	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegistration(BluetoothLeScanner.java:438)
[chromium] 	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:290)
[chromium] 	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:155)
[chromium] 	at mono.java.lang.RunnableImplementor.n_run(Native Method)
[chromium] 	at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:31)
[chromium] 	at android.os.Handler.handleCallback(Handler.java:938)
[chromium] 	at android.os.Handler.dispatchMessage(Handler.java:99)
[chromium] 	at android.os.Looper.loopOnce(Looper.java:201)
[chromium] 	at android.os.Looper.loop(Looper.java:288)
[chromium] 	at android.app.ActivityThread.main(ActivityThread.java:7870)
[chromium] 	at java.lang.reflect.Method.invoke(Native Method)
[chromium] 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
[chromium] 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
```
I guess I have to force the user to accept permission to scan ble somehow, right?

TimoKinnunen, 2022-11-24:
danijerez: This is older, but it can help you https://github.com/TimoKinnunen/MauiXAMLBluetoothLE. Android works fine.
I have issues with Windows.

janusw, 2022-11-25:
> With version 3.0.0-beta.2 in windows it seems that it already detects the devices correctly.

Awesome!

> I keep testing on Android 12.1, although I add the permission
> 
> ```
> <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
> ```
> 
> the following exception is thrown
> 
> [..]
> 
> I guess I have to force the user to accept permission to scan ble somehow, right?

On Android you have to request permissions at runtime, cf. https://learn.microsoft.com/en-us/dotnet/maui/platform-integration/appmodel/permissions?view=net-maui-7.0&tabs=android#requesting-permissions

TimoKinnunen, 2022-11-25:
You might be right that version 3.0.0-beta.2 in windows detects the devices correctly. But in Android smartphone after permissions are granted Bluetooth LE devices are shown, but not in Windows. A playground for you to see an example of asking permissions in Android (not perfect but functioning) is by opening in Visual Studio 2022 from GitHub https://github.com/TimoKinnunen/MauiXAMLBluetoothLE. Code in it is simple. Good luck to you.

TimoKinnunen, 2022-11-25:
My mistake!!!! I made a new test and it works.
Polar H7 Heart Rate Sensor (Bluetooth LE device) on Windows is detected by Plugin.BLE version 3.0.0-beta.2.
I am pleased. Thank you Janus Weil so far.
Android is ok. Windows is ok. iOS I don't know because I haven't tested it on my iPhone 10 yet. But I believe what you said that it is ok. Wonderful news.

TimoKinnunen, 2022-11-25:
Disconnecting Bluetooth LE device on Windows results in catastrophic failure. We have still work to do. But it has been amazing so far.

danijerez, 2023-01-29:
I kept doing tests. Version 3.0.0-beta.2 works fine for me on both android and windows. In android it sometimes lists unnamed devices and sometimes duplicates them. I did the tests with maui blazor, could the library be used in some way with windows forms or dotnet console? When is release 3.0 planned?

TimoKinnunen, 2023-01-29:
Thank you. It is wonderful to know that somebody has tested Plugin.BLE in a Maui app. I have no part in developing Plugin.BLE. Sorry.

janusw, 2023-02-02:
> I did the tests with maui blazor, could the library be used in some way with windows forms or dotnet console?

It's a pure netstandard2.0 / net6.0 library, and you should be able to use it independent of a UI framework. (I personally have neither tried it with Windows Forms nor Console, though.)

> When is release 3.0 planned?

We should release version 3.0 soon, I'll try to find some time for it in the coming weeks. But if beta.2 works well for you, you can just use that, anyway.

danijerez, 2023-02-03:
thanks for your answers, when I have tried it with windorms or dotnet console I get the following problem, I thought it was not compatible
`
This functionality is not implemented in the portable version of this assembly.  You should reference the NuGet package from your main application project in order to reference the platform-specific implementation.
`
How can I indicate the platform? I was testing it on windows

jamesmontemagno, 2023-02-05:
Official support is: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le#support--limitations

iOS, Android, macOS, and Windows with UWP or Win UI 3 and the Windows App SDK

This will not work with WinForms/WPF etc.

danijerez, 2023-04-10:
doing tests in windows 10, I get the following exception:

```
failed with status: AccessDenied
   at Plugin.BLE.Extensions.GattResultExtensions.ThrowIfError(GattCommunicationStatus status, String tag, Nullable`1 protocolError)
   at Plugin.BLE.Extensions.GattResultExtensions.ThrowIfError(GattCharacteristicsResult result, String tag)
   at Plugin.BLE.UWP.Service.GetCharacteristicsNativeAsync()
   at Plugin.BLE.Abstractions.ServiceBase`1.GetCharacteristicsAsync()
   at Plugin.BLE.Abstractions.ServiceBase`1.GetCharacteristicAsync(Guid id)
   at DigiKids.Services.BleSer
```

the implementation is simple:
```
public async Task<ICharacteristic> GetCharacteristic(IService service, Guid characteristicId)
{
    return await service.GetCharacteristicAsync(characteristicId);
}

```

in windows 11 this error is not reproduced, do I need any additional permission in 10?