dotnet-bluetooth-le/dotnet-bluetooth-le #597: [NO ISSUE WITH PLUGIN] Filter and for company id (advertisement data)

WanftMoon, 2022-05-18:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

Looking for a way to either filter (iOS and Android) for the manufacturer (0x0059), or check the advertisement records

1. Scan for devices from a given manufacturer (0x0059)

2. Device discovery, check advertisement data for company id

## Expected behavior
A way to check the company id of the discovered device

## Actual behavior
not sure how

### Crashlog

N/A

## Configuration

**Version of the Plugin:** vanilla version 2.1.3

**Platform:** iOS 10+ and Android 8+

**Device:** N/A

janusw, 2022-05-19:
> Looking for a way to either filter (iOS and Android) for the manufacturer (0x0059), or check the advertisement records

Both should be possible (at least on Android).

You can always check the ad records manually. Just subscribe to `DeviceAdvertised` to `DeviceDiscovered` (depending on your use case) and start scanning (as sketched here: https://github.com/xabre/xamarin-bluetooth-le#scan-for-devices). Then have a look at the `AdvertisementRecords` of the discovered devices.

About the filtering: Filtering on a manufacturer ID is only possible on Android (not iOS, but there you can still do it manually). Have a look at this: https://github.com/xabre/xamarin-bluetooth-le/blob/4b5a1065f0c17fae5a5b5890a59d69afe5e33b24/Source/Plugin.BLE.Abstractions/ScanFilterOptions.cs#L69

Simple usage example here: https://github.com/xabre/xamarin-bluetooth-le#scan-filtering

janusw, 2022-06-11:
I hope your question is answered. If not, please reply or open a new issue. I'll close this one for now.

closed 2022-06-11 by janusw